tejado/android-usb-gadget #4: Phone crashes (RESTARTS) After changing to keyboard and plugging in

yoyo00xx, 2020-09-16:
My kernal is linage kernal: 4.4
My phone is: HTC U11
I am rooted
Android 10

yoyo00xx, 2020-09-16:
How to replicate ? (Version 0.2 (latest))
1) Change to keyboard
2) Active
3) Plug in computer
4) Freezes + Restarts

kstuddard, 2020-09-23:
I have the same issue. I am running on a Pixel 2; I have experienced this with (rooted) stock 10 and 11, and Lineage 17.1
Same steps to reproduce; open app, press the + button and select "keyboard & mouse.sh", plug into PC. 
Phone then freezes, screen goes black, and it dies. I seem to have to wait a few seconds after unplugging to turn it back on. 

I would be very interested in actively helping resolve this issue but I don't really know where to start.

tejado, 2020-09-23:
Thanks a lot! I will have a look into it later today.

tejado, 2020-09-23:
Do you have a terminal app (like termux) installed on your phone? If yes, can you execute following command and provide the output? Thanks

`mount -t configfs`

kstuddard, 2020-09-23:
`none on /config type configfs (rw,nosuid,nodev,noexec,relatime)`

tejado, 2020-09-24:
@kstuddard 
Just to make sure: you never activated the added gadget by clicking the "Activate" button on the keyboard gadget or? At least this is missing in your description. I would assume, that the new keyboard gadget does not appear on the gadget list, as there are errors in your log when the gadget was (unsuccessful) added.

@yoyo00xx 
Can you provide me your log before you connect your PC to your smartphone?
Thank you!

kstuddard, 2020-09-25:
The keyboard gadget never appears for me to click "activate." There is only the /config/usb_gadget/g1/ gadget which I can activate or deactivate. I have tried selecting "mouse+keyboard.sh" with g1 already activated and deactivated, with no difference in outcome. 

Nothing I can do from the app seems to create the new gadgets appear in the app or make a difference in the outcome of `mount -t configfs` or `ls /config/usb_gadget/`

tejado, 2020-09-25:
@kstuddard 
Okay, this is what I expected when I read your logs. But very interessting that your phone freezes then because according to the log, you did not modify anything successful (all commands failed -> gadget was not added), except the deactivation of your standard gadget.

What will happen if you do the same without deactivating the standard gadget? Will the phone freeze again?

And can you try to create a folder "/config/usb_gadget/keyboard" manually (e.g. over a root shell) and check if there is any error (e.g. selinux) in logcat/dmesg due to this?

Thanks a lot for your help to debug this issue!

kstuddard, 2020-09-26:
from shell in the phone, running mkdir /config/usb_gadget/keyboard returns 
`mkdir: '/config/usb_gadget/keyboard': Out of memory`

I was working on a flowchart/table of sorts to see exactly when the crash occurs. 
My first set of tests had the Default Gadget (DG) enabled, KB+M not enabled, and the USB cable connected,. From here you can turn off & on DG without issue. Once you run the KB+M script it will enable DG if it's off already. After having ran KB+M, once you disable DG it will freeze (sometimes buzzing) and shut down the phone. 

My next set to test was having DG enabled already, having ran KB+M, and then plugging the phone in. After plugging the phone in the phone did not crash, but then I deactivated DG and the phone did crash. At this point it appeared to also fry the USB port on my laptop, so I am not going to continue this line of testing .

I did save logcats from this testing, but some of the earlier ones were done over USB before I realized how much more practical ADB over wifi would be while testing USB stuff. I also don't spend a lot of time in logcat so I am unsure how safe it is to just publish the contents. In the latter files I also tried to run `logcat -c` before starting the procedure in order to keep the logs smaller & cleaner. Give me a little bit and I'll figure out how to share these logcats and whether it's a safe idea to do so. That said, I grep'd specifically for SELinux and only found entries like this: 

```
./log3.txt:09-25 22:09:03.990  1241  1444 I SELinux : SELinux: Loaded file_contexts
./log5.txt:09-25 22:18:04.485  1241  1241 D SELinuxMMAC: Using policy file /system/etc/selinux/plat_mac_permissions.xml
./log5.txt:09-25 22:18:04.486  1241  1241 D SELinuxMMAC: Using policy file /vendor/etc/selinux/vendor_mac_permissions.xml
```
I failed to get any useful dmesg's at all.

SV133U, 2020-09-28:
Hi, I am also using the same methods in a rooted Cubot King Kong Mini phone and the mkdir keyboard command fails with Out of Memory and the phone resets! Looking for a solution to this problem I came up with your project (good) and this discussion. Seriously if you want I can give you access to a station with the phone connected and an adb shell for a debug session (but not a pizza ... Sorry!)

yoyo00xx, 2020-09-28:
`ROOT (stderr) - : <stdin>[53]: can't create strings/0x409/configuration: No such file or directory
ln: cannot create symbolic link from '/config/usb_gadget/keyboard/functions/hid.usb0' to '/config/usb_gadget/keyboard/configs/c.1//hid.usb0/hid.usb0': Operation not permitted
ln: cannot create symbolic link from '/config/usb_gadget/keyboard/functions/hid.usb1' to '/config/usb_gadget/keyboard/configs/c.1//hid.usb1/hid.usb1': Operation not permitted
ROOT - Execute command: for dir in /config/usb_gadget/*/; do echo GADGET_PATH=$dir; cd $dir; if [ "$?" -ne "0" ]; then echo "Error - not able to change dir to $dir... exit"; exit 1; fi; echo UDC=$(cat UDC); find ./configs/ -type l -exec sh -c 'echo FUNCTIONS_ACTIVE=$(basename $(readlink "$@"))' _ {} \;; for f in ./functions/*/; do echo FUNCTIONS=$(basename $f); done; cd ./strings/0x409/; for vars in *; do echo ${vars}=$(cat $vars); done; echo "============="; done; 
`

tejado, 2020-09-30:
Currently I see two issues which I need to verify if they are related or not:
1. Phone freeze when connecting it to a different device.
   During development, I had sometime the same issue but it was just due to a bad gadget configuration.
2. "Out Of Memory" error during folder creation in /config/usb_gadget/. 
   I don't have any idea why this happens. Maybe it is a bug in the kernel for specific versions.

@yoyo00xx 
Can you please provide the complete log? Or can you confirm that you have the same "Out of Memory" error message in your log?

@SV133U 
Wow, thank you! I really appreciate it. If this will be necessary, I come back to you. But maybe I will just buy the Cubot anyway.
Are you using LineageOS or any other custom rom as well? And if yes, what version exactly?

@kstuddard 
Yes, better to be save and control/check what logs you provide. There could be some sensitive content in it.
What happens if you only deactivate DG (Default Gadget) and plug your phone it? Does it freeze?

SV133U, 2020-10-02:
Hi! No LineageOS. To root the phone I followed the XDA guide and installed the AOSP 9 SU version already rooted. The kernel is version 4.9.117+
I had the idea of ​​installing Lineage, maybe I'll try ...

Jollastro, 2020-10-06:
Hi @tejado, i'm working with @SV133U with the Cubot king kong mini.

Here's a detailed explanation of what i did on the device and also the kernel log attached:

following this guide: https://forum.xda-developers.com/general/rooting-roms/installing-custom-rom-cubot-kingkong-t4050815

-Unlocked oem (in device setting)
-Rebooted the device on bootloader mode
-flashed:
    - fastboot flashing unlock
    - fastboot --disable-verity --disable-verification flash vbmeta vbmeta.img
    - fastboot flash boot boot.img (boot.img and vbmeta.img are from stock rom)
    - fastboot flash recovery recovery_TWRP.img (from the 1st comment)
    - fastboot flash lk lk_patched.img (from the 1st comment, just a booting red alert removal)
-rebooted to recovery twrp
-formatted data (format data, to remove encryption)
-rebooted the recovery after data format
-wiped system, dalvik, cache
-flashed AOSP 9 in system (already rooted)
-reboot to system

Everything is fine, device is rooted, recovery is still there and working (twrp) and aosp 9 works fine.

When i try to create a folder inside /config/usb_gadget/ i receive:
mkdir: can't create directory 'keyboard': Out of memory

If i disable the g1 gadget (utilising the usb_gadget app) and then plug in the phone nothing happends (no freeze).

Kernel log attached
[dmesg.log](https://github.com/tejado/android-usb-gadget/files/5332769/dmesg.log)

freeman1216, 2020-10-26:
Same for me. LineageOS 17.1 Samsung A40. "Out of memory" problem, also tried applying kernel patch, no good, no hdg device created

tejado, 2020-11-15:
@freeman1216 
Great, thanks for your update. I will add a feature to do this!

If anyone would like to still debug the "out of memory" issue, maybe it will help to strace the directory creation (strace bins can be found on the internet, e.g. https://github.com/ipduh/strace/tree/master/binaries )
I will need a new device for debugging & testing, this will take a while.

By the way: My kernel is 4.9 and there it is working.

tejado, 2020-12-21:
In v0.3 you can add functions to existing gadgets.
Please check if this is working for you:
https://github.com/tejado/android-usb-gadget/releases/tag/v0.3-beta

tejado, 2020-12-22:
Additionally: I had some time to analyze the issue. It seems that on specific kernel versions, the internal device is created by configfs always as "android0". Adding further gadgets is not possible as android0 already exists.

So please try to add Mouse/Keyboard functions to the existing gadget, as @freeman1216 described. This is now possible with android-usb-gadget v0.3.
Can you please test this, @Jollastro @freeman1216 @SV133U @kstuddard @yoyo00xx  ?

References:
https://e2e.ti.com/support/processors/f/791/t/598656?J6EVM5777-Adding-USB-Gadget-on-J6-Android-OS-with-NCM-support
https://github.com/corsicanu/android_kernel_samsung_universal8895/blob/9/drivers/usb/gadget/configfs.c#L2041

yoyo00xx, 2020-12-26:
it did not crash and it added the functions but i was not able to test it. For example Rucky does not detect USB gadget. If you could provide me with the linux commands I can test in the shell if I can send keystrokes it would be great @tejado

Jollastro, 2020-12-28:
Same here, adding the functions is working, no crashes.
But once the functions are added i'm unable to identify the devices (/dev/hid0 or /dev/hid1) to utilize the hid-gadget-test.

Another point is: how should the hid-gadget-test be used? Because the "usage print" in the terminal says one way of utilizing it and the README in the github page says another way.

tejado, 2020-12-28:
I have the same issue and currently check what options are there. 
For one of my devices, deactivating all other functions worked. But at a second device, the gadget gets deactivated when ADB function will be deactivated.

The issue is basically inside the kernel... few are working well but a lot of other vendor kernels are not. Depending on the patch level... :/

@Jollastro 
hid-gadget-test ? This is not part of this project, so I'm not sure what pages you are referring to.

Jollastro, 2020-12-29:
The tool i'm talking about is linked in your use-cases (https://github.com/pelya/android-keyboard-gadget).

I'm trying to use it to send keystrokes to the pc with the smartphone but i can't find the device path inside the /dev folder.
I've added the keyboard and mouse functions to the default g1 gadget and it doesn't crash but i can't find the devices it creates.

tejado, 2020-12-29:
If /dev/hid* devices are not present, the function is not active. See dmesg (e.g. run dmesg -w parallel) during adding the function, activating the gadget and so on for more information. There you see the kernel messages which gives some information why it fails.

Jollastro, 2021-03-01:
Hi tejado, /dev/hid1 or /dev/hid2 are not present. Here is the output of dmesg -w parallel while adding the mouse function:

[image]

Jollastro, 2021-03-04:
And this is the log of adding the keyboard function:
[image]

Kosaci, 2021-09-21:
> I have the same issue. I am running on a Pixel 2; I have experienced this with (rooted) stock 10 and Lineage 17.1
> Same steps to reproduce; open app, press the + button and select "keyboard & mouse.sh", plug into PC.
> Phone then freezes, screen goes black, and it dies. I seem to have to wait a few seconds after unplugging to turn it back on.
> 
> I would be very interested in actively helping resolve this issue but I don't really know where to start.
> 
> Here is the log; I just open the app, deactivate, and add the kb&m.
> 
> `ROOT - Execute command: for dir in /config/usb_gadget/_/; do echo GADGET_PATH=$dir; cd $dir; if [ "$?" -ne "0" ]; then echo "Error - not able to change dir to $dir... exit"; exit 1; fi; echo UDC=$(cat UDC); find ./configs/ -type l -exec sh -c 'echo FUNCTIONS_ACTIVE=$(basename $(readlink "$@"))' _ {} ;; for f in ./functions/_/; do echo FUNCTIONS=$(basename $f); done; cd ./strings/0x409/; for vars in *; do echo ${vars}=$(cat $vars); done; echo "============="; done;
> 
> # ROOT (stdout) - GADGET_PATH=/config/usb_gadget/g1/
> UDC=a800000.dwc3
> FUNCTIONS_ACTIVE=ffs.adb
> FUNCTIONS=accessory.gs2
> FUNCTIONS=audio_source.gs3
> FUNCTIONS=cser.dun.0
> FUNCTIONS=cser.nmea.1
> FUNCTIONS=diag.diag
> FUNCTIONS=ffs.adb
> FUNCTIONS=ffs.mtp
> FUNCTIONS=ffs.ptp
> FUNCTIONS=gsi.rmnet
> FUNCTIONS=gsi.rndis
> FUNCTIONS=mass_storage.0
> FUNCTIONS=midi.gs5
> FUNCTIONS=qdss.qdss
> manufacturer=Google
> product=Pixel 2
> serialnumber=HT7AL1A02668
> ROOT (stderr) -
> ROOT - Execute command: echo "" > /config/usb_gadget/g1//UDC
> 
> ROOT (stdout) -
> ROOT (stderr) -
> ROOT - Execute command: for dir in /config/usb_gadget/_/; do echo GADGET_PATH=$dir; cd $dir; if [ "$?" -ne "0" ]; then echo "Error - not able to change dir to $dir... exit"; exit 1; fi; echo UDC=$(cat UDC); find ./configs/ -type l -exec sh -c 'echo FUNCTIONS_ACTIVE=$(basename $(readlink "$@"))' _ {} ;; for f in ./functions/_/; do echo FUNCTIONS=$(basename $f); done; cd ./strings/0x409/; for vars in *; do echo ${vars}=$(cat $vars); done; echo "============="; done;
> 
> # ROOT (stdout) - GADGET_PATH=/config/usb_gadget/g1/
> UDC=
> FUNCTIONS_ACTIVE=ffs.adb
> FUNCTIONS=accessory.gs2
> FUNCTIONS=audio_source.gs3
> FUNCTIONS=cser.dun.0
> FUNCTIONS=cser.nmea.1
> FUNCTIONS=diag.diag
> FUNCTIONS=ffs.adb
> FUNCTIONS=ffs.mtp
> FUNCTIONS=ffs.ptp
> FUNCTIONS=gsi.rmnet
> FUNCTIONS=gsi.rndis
> FUNCTIONS=mass_storage.0
> FUNCTIONS=midi.gs5
> FUNCTIONS=qdss.qdss
> manufacturer=Google
> product=Pixel 2
> serialnumber=HT7AL1A02668
> ROOT (stderr) -
> ROOT - Execute command: #!/bin/sh
> 
> CONFIGFS_DIR="/config"
> GADGETS_PATH="${CONFIGFS_DIR}/usb_gadget"
> 
> GADGET="keyboard"
> GADGET_PATH=${GADGETS_PATH}/${GADGET}
> 
> CONFIG_PATH="$GADGET_PATH/configs/c.1/"
> STRINGS_PATH="$GADGET_PATH/strings/0x409/"
> 
> mkdir -p $CONFIG_PATH
> mkdir -p $STRINGS_PATH
> 
[... 78 lines cut ...]
> FUNCTIONS=accessory.gs2
> FUNCTIONS=audio_source.gs3
> FUNCTIONS=cser.dun.0
> FUNCTIONS=cser.nmea.1
> FUNCTIONS=diag.diag
> FUNCTIONS=ffs.adb
> FUNCTIONS=ffs.mtp
> FUNCTIONS=ffs.ptp
> FUNCTIONS=gsi.rmnet
> FUNCTIONS=gsi.rndis
> FUNCTIONS=mass_storage.0
> FUNCTIONS=midi.gs5
> FUNCTIONS=qdss.qdss
> manufacturer=Google
> product=Pixel 2
> serialnumber=HT7AL1A02668
> ROOT (stderr) -
> `

1. First try to create directory of keyboard(hid.keyboard) or mouse  in /config/usb_gadget/g1/function.
2. Enter the directory of keyboard(cd hid.keyboard).
3. Check the output of report_desc(cat report_desc).
4. If the files gives any output then it is alredy configured for keyboard and make symlink at ../config/b.1/hid.keyboard.
5. Finally activate at usb_gadget_tool.
6. It creates hidg* files.

aka56kg, 2022-10-05:
>     4. If the files gives any output then it is alredy configured for keyboard and make symlink at ../config/b.1/hid.keyboard.

How do you create symlinks? I don't understand, can you give me a hint?

aka56kg, 2022-10-12:
> 1. First try to create directory of keyboard(hid.keyboard) or mouse  in /config/usb_gadget/g1/function.
>
>     2. Enter the directory of keyboard(cd hid.keyboard).
>
>     3. Check the output of [...]
Doesn't work for me, usb keyboard app still doesn't work after this, also, after reboot, these folders are getting deleted

Kosaci, 2022-10-14:
What happened after creating directory...