tejado/android-usb-gadget #5: There was a problem while parsing the package

bgeke, 2020-10-11:
I tried both release apks on Moto G (falcon) running rooted Lineage os 14.1 and Redmi note4 (mido) running stock MIUI but getting error "There was a problem while parsing the package" while trying to open apk. I understand kernels of these devices are old but I don't know why this error is coming? Excuse if I don't know basic things.

I tried same apk in stock Redmi Note7S and it opens. just saying the app requires root to work. I know this is for rooted devices and just given a try.

tejado, 2020-11-15:
Hi @bgeke 
please enable "Allow installation of apps from unknown sources" in your android settings. If this is enabled and you get still this issue, it is maybe caused due to the new way of signing android apks.
And can you please check what kernel are you using?

zzzop, 2020-11-28:
same error on redmi 1s lineageos 14.1 android 7.1.2 kernel 3.4.0

goshil, 2020-12-04:
Same error, unknown sources enabled, android 6.0.1 , xnano x5 tv box, kernel 4.1.17

remixer-dec, 2021-04-10:
Looks like the app [has](https://github.com/tejado/android-usb-gadget/blob/master/app/build.gradle#L8) minSdkVersion 26, which means you can only install it on Android >= [8](https://developer.android.com/studio/releases/platforms#8.0)

DRSDavidSoft, 2021-04-28:
Hopefully, the lower SDKs would also be supported (e.g. mine is SDK25, in #5)

tejado, 2021-04-29:
You would need at least a kernel in version 3.19 or higher. 

@goshil is there no ROM available with a more up to date Android version?

goshil, 2021-04-29:
> @goshil is there no ROM available with a more up to date Android version?

No, just Android 6 ...

goshil, 2021-04-29:
I've recompiled app 0.3-beta with **minSdkVersion 26** , so try out this apk.
[app-debug.zip](https://github.com/tejado/android-usb-gadget/files/6398960/app-debug.zip)

tejado, 2021-04-29:
Does it work for you? 
Feel free to make a PR to decease the minSdkVersion.

goshil, 2021-04-29:
> Does it work for you?
> Feel free to make a PR to decease the minSdkVersion.

No, it does not even starting - crashing at startup. But my kernel is >4 version, i would like to try it out at least.

tejado, 2021-04-29:
I will check later today if I can decrease the version for you.

feboyfierlyan, 2021-05-04:
> I will check later today if I can decrease the version for you.

have you decrease the version?

tejado, 2021-05-10:
I set minSdkVersion to 25. Can you please try if this one will work for you? Just for testing. If this works, I will make a separate release.
[app-release.zip](https://github.com/tejado/android-usb-gadget/files/6453899/app-release.zip)

goshil, 2021-05-15:
> I set minSdkVersion to 25. Can you please try if this one will work for you? Just for testing. If this works, I will make a separate release.
> [app-release.zip](https://github.com/tejado/android-usb-gadget/files/6453899/app-release.zip)

Still "here was a problem while parsing the package" on installation.
My version could be installed, at least...

goshil, 2021-05-15:
App with minSDK 23 : https://github.com/tejado/android-usb-gadget/files/6398960/app-debug.zip

tejado, 2021-05-16:
@goshil 
So you need minSdkVersion 23? Is your app working? Or is your kernel too old? ;)

goshil, 2021-05-16:
> @goshil
> So you need minSdkVersion 23? Is the app working now? Or is your kernel too old? ;)

No... Installation successful, but app crashes on startup...

DRSDavidSoft, 2021-05-21:
@tejado It's possible to build a custom kernel for my device, I'm looking into this and will report back.

tejado, 2021-06-03:
SDK version is decreased to 23 in the latest release: https://github.com/tejado/android-usb-gadget/releases/tag/v0.4-beta

closed 2021-06-03 by tejado