tejado/android-usb-gadget #6: Is it possible to use this app as a library?

TheBricktop, 2020-10-11:
I want to prototype a device  that acts as a mouse and control it from opencv app , for my purpose i need to build it as a aar lib, is it possible to build such binary or there is an easier way to it?

tejado, 2020-11-15:
You can get all the details from the source. So you can use it partly as a lib but there is no documentation on it except my source code.
But android-usb-gadget just enables you to use your Android phone as a mouse/keyboard. You can still implement the device logic, so acting like a mouse/keyboard by writing to /dev/hidg* by any other code, e.g. your aar lib. This code is provided here:
https://github.com/tejado/Authorizer/tree/master/authorizer/src/main/java/net/tjado/authorizer

I'm curious: what is the use-case of your app? :-)

TheBricktop, 2020-11-16:
I'm trying to convert opencv processed image to hid mouse, and then go for bluetooth hid.

tejado, 2020-11-16:
This sounds interesting. For keyboard, you need to convert  from the images to characters and then you can get the respective HID scancodes by mapping them (e.g. https://github.com/tejado/Authorizer-PoC/blob/master/app/src/main/java/net/tjado/authorizer/UsbHidKbd_en_US.java). These scancodes can then be used in USB and Bluetooth.

TheBricktop, 2020-11-19:
Yea I have an idea how to get the parameters (x y and button click) but the boilerplate code without any ui dependency for bluetoothhidprofile gets me confused as there is no official documentation on this or im not googly enough to find it.
Basically my library should need at least two functions one for starting the bluetooth client with hid profile and then one for sending the report with params (int x int y bool clicked).

tejado, 2020-11-19:
Look up BluetoothHidDevice and how it is used in my code or other sample apps, e.g. https://github.com/ralismark/bluehid.
You have to check how to use it without any UI because you will need some way of pairing and so on (which is done by Android but you have to approve it and so on).

TheBricktop, 2020-11-20:
Yea im currently hacking ralismark's approach but got stuck on the mBtHidDevice.registerApp cause it needs a executor and Im not sure if i should somehow get the executor from the  unitymainActivity (cause my intetntion is to use this lib in unity) or create it on the android side with context provided from the unity. Its largely convoluted and im kinda stuck.