tejado/android-usb-gadget #8: MacOS 11 cant detect keyboard type

totaleo, 2020-12-13:
Hi. I'm install your app on LinageOS 17 + magisk 21 as root, then connect phone by usb-c to my mac laptop and have keyboard configuration window like that:
https://drive.google.com/file/d/129ooWJUkLOKHk0sJt_79gRnaUnVIN8uj/view?usp=sharing

The are should be something wrong with keyboard identifiers. Probably will have same result on previous MacOS versions.

I cant provide any additions logs by your request.

tejado, 2020-12-13:
Thanks for reporting it.
Seems Apple is a little bit picky: https://forum.yubico.com/viewtopic82f0.html?f=4&t=111
I will check what the best cross-platform identifies could be.

totaleo, 2020-12-17:
Yes. The problem is because MacOS can't detect the VID/PID provided by current app.
I found a website with known VIP/PID values for different manufactures:
https://ducktoolkit.com/vidpid

tejado, 2020-12-21:
Please check if the new release is working for you!
 https://github.com/tejado/android-usb-gadget/releases/tag/v0.3-beta

totaleo, 2020-12-27:
Hi. Still is not detectable on MacOS. I tested on 11.1 and 10.15.

I used this parameters for MacOS, works fine on different OS versions:

```
# HID/VID
echo 0x05ac > idVendor
echo 0x0201 > idProduct

mkdir -p strings/0x409
echo "Apple, Inc." > strings/0x409/manufacturer
echo "USB Keyboard" > strings/0x409/product
echo "0123456789" > strings/0x409/serialnumber
```

totaleo, 2020-12-27:
But this is works only on MacOS. I dont have ability to check how is Windows machines will works with this.
I did try to use Windows on VM to check, but not sure that VM software will pass correct params from host to guest.

alex9099, 2021-03-22:
I just knew about this and it's awesome that this userland api came precompiled in the kernel I'm using.

But the keyboard doesn't seem to be recognized on an "old" computer bios (think it's american megatrends 2002 or smth like that), perhaps it can't even use usb keyboards, I'm not sure as I don't have a keyboard with me (thus using this application)

On my linux laptop it gets recognized without a problem