tejado/android-usb-gadget #12: out of memory

dodoradio, 2021-04-14:
I am running lineageos 18.1 (android 11 with magisk root) on a samsung note 9 N-960f. I was met with the same issue with OneUI2.5 rom (android 10 with magisk root). I have downloaded the source and tried manually executing the shellscripts for gadget creation but it gives the same 'out of memory' error. I also tried manually creating the directories within /config/usb_gadget and i get the same 'out of memory' error.  
```
ROOT (stdout): GADGET_PATH=/config/usb_gadget/g1/
    CONFIG_PATH=/config/usb_gadget/g1//configs/b.1/
    UDC=10c00000.dwc3
    FUNCTIONS_ACTIVE=ffs.adb
    FUNCTIONS_ACTIVE=ncm.0
    FUNCTIONS_ACTIVE=audio_source.0
    FUNCTIONS_ACTIVE=accessory.0
    FUNCTIONS_ACTIVE=midi.0
    FUNCTIONS_ACTIVE=conn_gadget.0
    FUNCTIONS_ACTIVE=dm.0
    FUNCTIONS_ACTIVE=rndis.0
    FUNCTIONS_ACTIVE=acm.0
    FUNCTIONS_ACTIVE=ffs.adb
    FUNCTIONS_ACTIVE=ptp.0
    FUNCTIONS_ACTIVE=mtp.0
    FUNCTIONS=accessory.0
    FUNCTIONS=acm.0
    FUNCTIONS=audio_source.0
    FUNCTIONS=conn_gadget.0
    FUNCTIONS=dm.0
    FUNCTIONS=ffs.adb
    FUNCTIONS=midi.0
    FUNCTIONS=mtp.0
    FUNCTIONS=ncm.0
    FUNCTIONS=ptp.0
    FUNCTIONS=rndis.0
    manufacturer=SAMSUNG
    product=SAMSUNG_Android
    serialnumber=27681114320d7ece
    =============
2021-04-14 09:34:54.227 1482-1482/net.tjado.usbgadget W/tjado.usbgadge: Accessing hidden method Landroid/graphics/FontFamily;-><init>()V (greylist-max-q, reflection, allowed)
2021-04-14 09:34:54.227 1482-1482/net.tjado.usbgadget W/tjado.usbgadge: Accessing hidden method Landroid/graphics/FontFamily;->addFontFromAssetManager(Landroid/content/res/AssetManager;Ljava/lang/String;IZIII[Landroid/graphics/fonts/FontVariationAxis;)Z (greylist-max-q, reflection, allowed)
2021-04-14 09:34:54.227 1482-1482/net.tjado.usbgadget W/tjado.usbgadge: Accessing hidden method Landroid/graphics/FontFamily;->addFontFromBuffer(Ljava/nio/ByteBuffer;I[Landroid/graphics/fonts/FontVariationAxis;II)Z (greylist-max-q, reflection, allowed)
2021-04-14 09:34:54.227 1482-1482/net.tjado.usbgadget W/tjado.usbgadge: Accessing hidden method Landroid/graphics/FontFamily;->freeze()Z (greylist-max-q, reflection, allowed)
2021-04-14 09:34:54.227 1482-1482/net.tjado.usbgadget W/tjado.usbgadge: Accessing hidden method Landroid/graphics/FontFamily;->abortCreation()V (greylist-max-q, reflection, allowed)
2021-04-14 09:34:54.227 1482-1482/net.tjado.usbgadget W/tjado.usbgadge: Accessing hidden method Landroid/graphics/Typeface;->createFromFamiliesWithDefault([Landroid/graphics/FontFamily;Ljava/lang/String;II)Landroid/graphics/Typeface; (greylist, reflection, allowed)
2021-04-14 09:34:58.352 1482-8452/net.tjado.usbgadget I/mali_winsys: new_window_surface() [1725x926] return: 0x3000
2021-04-14 09:35:01.039 1482-8470/net.tjado.usbgadget D/ROOT: Execute command: #!/bin/sh
    
    CONFIGFS_DIR="/config"
    GADGETS_PATH="${CONFIGFS_DIR}/usb_gadget"
    
    GADGET="keyboard"
    GADGET_PATH=${GADGETS_PATH}/${GADGET}
    
    CONFIG_PATH="$GADGET_PATH/configs/c.1/"
    STRINGS_PATH="$GADGET_PATH/strings/0x409/"
    
    mkdir -p $CONFIG_PATH
    mkdir -p $STRINGS_PATH
    
    mkdir -p $GADGET_PATH/functions/hid.keyboard
    cd $GADGET_PATH/functions/hid.keyboard
    
    # HID protocol (according to USB spec: 1 for keyboard)
    echo 1 > protocol
    # device subclass
    echo 1 > subclass
    # number of bytes per record
    echo 8 > report_length
    
    # writing report descriptor
    echo -ne \\x05\\x01\\x09\\x06\\xa1\\x01\\x05\\x07\\x19\\xe0\\x29\\xe7\\x15\\x00\\x25\\x01\\x75\\x01\\x95\\x08\\x81\\x02\\x95\\x01\\x75\\x08\\x81\\x03\\x95\\x05\\x75\\x01\\x05\\x08\\x19\\x01\\x29\\x05\\x91\\x02\\x95\\x01\\x75\\x03\\x91\\x03\\x95\\x06\\x75\\x08\\x15\\x00\\x25\\x65\\x05\\x07\\x19\\x00\\x29\\x65\\x81\\x00\\xc0 > report_desc
    
    mkdir -p $GADGET_PATH/functions/hid.mouse
    cd $GADGET_PATH/functions/hid.mouse
    
    # HID protocol (according to USB spec: 2 for mouse)
    echo 2 > protocol
    # device subclass
    echo 1 > subclass
    # number of bytes per record
    echo 4 > report_length
    
    # writing report descriptor
    echo -ne \\x05\\x01\\x09\\x02\\xa1\\x01\\x09\\x01\\xa1\\x00\\x05\\x09\\x19\\x01\\x29\\x05\\x15\\x00\\x25\\x01\\x95\\x05\\x75\\x01\\x81\\x02\\x95\\x01\\x75\\x03\\x81\\x01\\x05\\x01\\x09\\x30\\x09\\x31\\x09\\x38\\x15\\x81\\x25\\x7F\\x75\\x08\\x95\\x03\\x81\\x06\\xc0\\xc0 > report_desc
    
    
    cd $GADGET_PATH
    echo 0x046a > idVendor
    echo 0x002a > idProduct
    
    cd $STRINGS_PATH
    echo "tejado" > manufacturer
    echo "Android USB Gadget" > product
    echo "42" > serialnumber
    
    cd $CONFIG_PATH
    echo 120 > MaxPower
    echo "Configuration" > strings/0x409/configuration
    
    ln -s ${GADGET_PATH}/functions/hid.keyboard $CONFIG_PATH/hid.keyboard
    ln -s ${GADGET_PATH}/functions/hid.mouse $CONFIG_PATH/hid.mouse
2021-04-14 09:35:01.041 1482-8452/net.tjado.usbgadget I/mali_egl: eglDestroySurface() in
2021-04-14 09:35:01.043 1482-8452/net.tjado.usbgadget I/mali_winsys: delete_surface() [1725x926] return
2021-04-14 09:35:01.043 1482-8452/net.tjado.usbgadget I/mali_egl: eglDestroySurface() out
2021-04-14 09:35:01.043 1482-8452/net.tjado.usbgadget W/libEGL: EGLNativeWindowType 0x75155f0950 disconnect failed
2021-04-14 09:35:01.043 1482-8452/net.tjado.usbgadget D/OpenGLRenderer: endAllActiveAnimators on 0x74955cc890 (AlertController$RecycleListView) with handle 0x73b5620530
2021-04-14 09:35:01.119 1482-8470/net.tjado.usbgadget D/ROOT (stderr): mkdir: '/config/usb_gadget/keyboard': Out of memory
    mkdir: '/config/usb_gadget/keyboard': Out of memory
    mkdir: '/config/usb_gadget/keyboard': Out of memory
    : <stdin>[16]: cd: /config/usb_gadget/keyboard/functions/hid.keyboard: No such file or directory
    : <stdin>[19]: can't create protocol: Read-only file system
    : <stdin>[21]: can't create subclass: Read-only file system
    : <stdin>[23]: can't create report_length: Read-only file system
    : <stdin>[26]: can't create report_desc: Read-only file system
    mkdir: '/config/usb_gadget/keyboard': Out of memory
```

dodoradio, 2021-04-14:
this log was taken out of an adb shell so i could create the issue on my laptop

CyberKid1987, 2021-04-27:
We try a lot of times with Nethunter team to fix this problem. I Have galaxy note 9 and try it on dr ketans roms (9,10) and lineage 17.1. Fuckin exynos! I post the problem for not fixing. We try patched and not patched kernel with few init.nethunter.rc files. 

[image]
[image]

yesimxev, 2021-04-28:
Yes the mistery we need to solve is this: why is kernel settings like CONFIGFS_F_HID is not effective? We've set it to =y and doesn't happen, although the kernel is flashed, everything else works as expected after our usual patches. I know it's not really an app issue, but you might have a clue @tejado

dodoradio, 2021-04-28:
do you mean some kind of hardware limitation by 'samsung limitation'?

CyberKid1987, 2021-04-28:
> do you mean some kind of hardware limitation in the exynos chip by 'samsung limitation'?

Perchaps

CyberKid1987, 2021-04-28:
@krzk we need your help.

krzk, 2021-04-28:
CONFIGFS_F_HID is a kernel configuration option, so it's not connected with anything in hardware. Why is it not enabled? You need to check your kernel config. It's difficult to help as I don't know which kernel are you using.

The initial report about out of memory looks like user-space problem. To be sure - you can run "dmesg" on the device or try to get kernel logs from adb. The OOM-killer or page allocation failures (when there is not enough of memory for Linux kernel) should be reported.

krzk, 2021-04-28:
Looking slightly more - here out of memory seems to be after _mkdir: '/config/usb_gadget/keyboard'_. What type of file system is /config? configfs (USB_CONFIGFS_F_HID depends on it)? Or maybe it's something specific to Android?

Just to clarify - I maintain the Samsung Exynos SoC for mainline Linux kernel, which is my hobby. If you use Linux kernel coming from a market device - so called vendor or Samsung Android kernel - it deviates from mainline by ~1 - 2 millions of lines of code. It's basically entirely different... That code is not maintained by anyone - neither by me, nor by Samsung.

dodoradio, 2021-04-28:
so i am currently running nezukoOS built on this kernel https://github.com/synt4x93/android_kernel_samsung_universal9810, and obviously it does not work.

krzk, 2021-04-28:
That's Exynos9810, so definitely a vendor kernel. Also commit "import N960FXXU5DTCA OSRC" points to it. It can work, it might not. It might be buggy, might not. Might be doing things entirely different than Linux kernel (`#define True False`)... Who knows?

Support is only for the things in the upstream. If someone decides not to upstream source code, he/she needs to live with the outcome.

tejado, 2021-04-29:
@krzk nice to meed you :-)

/config is the [USB Gadget ConfigFS filesystem](https://www.kernel.org/doc/html/latest/usb/gadget_configfs.html) - so not Android specific.

We have two issues here:
1.  CONFIGFS_F_HID is disabled. This is just a minor issue and can be resolved by a.) getting a kernel with this flag enabled or b) implementing it in userspace. I'm working on a userspace library for it but it is a lot of work due to the instability and less documentation of this part of the kernel. @CyberKid1987 

2. "out of memory" (spoiler: it is not a real OOM issue) - and here I could really need your help @krzk. I'm not a kernel dev so it is hard for me to find the real cause. Mostly Samsung devices and few others (see https://github.com/tejado/android-usb-gadget/issues/4) are affected on this. Pixel 3 worked perfect for me. The underlying issue seems to be inside the Android modifications of the kernel. And there are two types of problems I could identify:
    1. hardcoded gadget enumeration (at [this LoC](https://android.googlesource.com/kernel/msm/+/android-7.1.0_r0.2/drivers/usb/gadget/configfs.c?autodive=0%2F%2F%2F%2F#1703) you see the issue) - can be circumvented by using the default system gadget (normally "/config/usb_gadget/g1/") and not creating a new one. Adding Functions to existing Gadgets was a new feature I introduced in the last version due to this. Newer kernel versions (depending on the vendor) fixed that as well (android0 -> android%d).
    2. But using now the default system gadget (g1) does not let me really use ffs.hid or userspace functions. There are some hard dependencies to ADB which seems to cause conflicts with the underlying controller, e.g. hard disabling ffs.adb in the gadget removes the gadget<->controller binding (-> UDC empty). And using ffs.adb with ffs.hid together does not work (/dev/hidg* device is not created).

I'm happy to look together on this issue. I have more deeper kernel logs on this (e.g. by adding more debug messages to /drivers/usb/gadget/configfs.c and co) but there were not helpful for me. Any hint is highly appreciated.

krzk, 2021-04-29:
I am afraid the only reasonable hint I can give is to dump this kernel from Samsung. It's full of hacks, not-upstreamed code and pieces of questionable quality. More over, it differs from the mainline so whatever I would say, might not be applicable here. Unfortunately if you ever use vendor kernel, you are entirely on your own (assuming vendor does not care).

tejado, 2021-04-29:
Okay, thanks but I will not dump it ;) Samsung is not the only Kernel with this issue, so I want to know the cause of the ffs.adb conflict and if there is a chance to resolve it (even with a kernel patch).

CyberKid1987, 2021-04-29:
> @krzk nice to meed you :-)
> 
> /config is the [USB Gadget ConfigFS filesystem](https://www.kernel.org/doc/html/latest/usb/gadget_configfs.html) - so not Android specific.
> 
> We have two issues here:
> 
>     1. CONFIGFS_F_HID is disabled. This is just a minor issue and can be resolved by a.) getting a kernel with this flag enabled or b) implementing it in userspace. I'm working on a userspace library for it but it is a lot of work due to the instability and less documentation of this part of the kernel. @CyberKid1987
> 
>     2. "out of memory" (spoiler: it is not a real OOM issue) - and here I could really need your help @krzk. I'm not a kernel dev so it is hard for me to find the real cause. Mostly Samsung devices and few others (see #4) are affected on this. Pixel 3 worked perfect for me. The underlying issue seems to be inside the Android modifications of the kernel. And there are two types of problems I could identify:
>        
>        1. hardcoded gadget enumeration (at [this LoC](https://android.googlesource.com/kernel/msm/+/android-7.1.0_r0.2/drivers/usb/gadget/configfs.c?autodive=0%2F%2F%2F%2F#1703) you see the issue) - can be circumvented by using the default system gadget (normally "/config/usb_gadget/g1/") and not creating a new one. Adding Functions to existing Gadgets was a new feature I introduced in the last version due to this. Newer kernel versions (depending on the vendor) fixed that as well (android0 -> android%d).
>        2. But using now the default system gadget (g1) does not let me really use ffs.hid or userspace functions. There are some hard dependencies to ADB which seems to cause conflicts with the underlying controller, e.g. hard disabling ffs.adb in the gadget removes the gadget<->controller binding (-> UDC empty). And using ffs.adb with ffs.hid together does not work (/dev/hidg* device is not created).
> 
> 
> I'm happy to look together on this issue. I have more deeper kernel logs on this (e.g. by adding more debug messages to /drivers/usb/gadget/configfs.c and co) but they were not helpful for me.
> Any hint or idea is highly appreciated.

1) yesimxev fix it but not working hid 🙂

tejado, 2021-04-29:
>     1. yesimxev fix it but not working hid 🙂

Sorry, what do you mean? :-) Is the kernel config now showing that it is enabled?

CyberKid1987, 2021-04-29:
[image]
[image]

CyberKid1987, 2021-04-29:
> > ```
> > 1. yesimxev fix it but not working hid 🙂
> > ```
> 
> Sorry, what do you mean? :-) Is the kernel config now showing that it is enabled?

Yes, you can see on picture.
Another method but Fail again. 
[image]

tejado, 2021-04-29:
Okay, can you provide more info? Did you add then ffs.hid to the gadget (app log + dmesg log) ? And if this worked, is ffs.adb still enabled on the same gadget or not (did you created a separate gadget?) ?

krzk, 2021-04-29:
> 2. But using now the default system gadget (g1) does not let me really use ffs.hid or userspace functions. There are some hard dependencies to ADB which seems to cause conflicts with the underlying [...]

The DWC USB controller itself does not care, neither does driver in mainline kernel. The vendor could do many things with it, including disabling of specific feature etc.

CyberKid1987, 2021-04-29:
> Okay, can you provide more info? Did you add then ffs.hid to the gadget (app log + dmesg log) ? And if this worked, is ffs.adb still enabled on the same gadget or not (did you created a separate gadget?) ?

[image]

CyberKid1987, 2021-04-29:
Try to add this repos but same, not working. I mean usb, permissive and config. 

https://gitea.com/fasheng/magisk-modules

CyberKid1987, 2021-04-29:
ROOT - Execute command: for dir in /config/usb_gadget/*/; do echo GADGET_PATH=$dir; cd $dir/configs/; echo CONFIG_PATH="$dir/configs/`ls -1 | head -1`/"; cd $dir; if [ "$?" -ne "0" ]; then echo "Error - not able to change dir to $dir... exit"; exit 1; fi; echo UDC=$(cat UDC); find ./configs/ -type l -exec sh -c 'echo FUNCTIONS_ACTIVE=$(basename $(readlink "$@"))' _ {} \;; for f in ./functions/*/; do echo FUNCTIONS=$(basename $f); done; cd ./strings/0x409/; for vars in *; do echo ${vars}=$(cat $vars); done; echo "============="; done; 

ROOT (stdout) - GADGET_PATH=/config/usb_gadget/g1/
CONFIG_PATH=/config/usb_gadget/g1//configs/b.1/
UDC=10c00000.dwc3
FUNCTIONS_ACTIVE=ncm.0
FUNCTIONS_ACTIVE=audio_source.0
FUNCTIONS_ACTIVE=accessory.0
FUNCTIONS_ACTIVE=midi.0
FUNCTIONS_ACTIVE=conn_gadget.0
FUNCTIONS_ACTIVE=dm.0
FUNCTIONS_ACTIVE=rndis.0
FUNCTIONS_ACTIVE=acm.0
FUNCTIONS_ACTIVE=ffs.adb
FUNCTIONS_ACTIVE=ptp.0
FUNCTIONS_ACTIVE=mtp.0
FUNCTIONS=accessory.0
FUNCTIONS=acm.0
FUNCTIONS=acm.usb0
FUNCTIONS=audio_source.0
FUNCTIONS=conn_gadget.0
FUNCTIONS=dm.0
FUNCTIONS=ecm.usb0
FUNCTIONS=ffs.adb
FUNCTIONS=hid.0
FUNCTIONS=hid.1
FUNCTIONS=midi.0
FUNCTIONS=mtp.0
FUNCTIONS=ncm.0
FUNCTIONS=ptp.0
FUNCTIONS=rndis.0
manufacturer=SAMSUNG
product=SAMSUNG_Android
serialnumber=27c9f8a185217ece
=============

ROOT (stderr) - 
ROOT - Execute command: echo KERNEL_VERSION=`(uname -r |cut -d '-' -f1 )` && (gunzip -c /proc/config.gz | grep -i configfs | sed 's/# //; s/ is not set/=NOT_SET/')

ROOT (stdout) - KERNEL_VERSION=4.9.218
CONFIG_USB_CONFIGFS=y
CONFIG_USB_CONFIGFS_SERIAL=NOT_SET
CONFIG_USB_CONFIGFS_ACM=y
CONFIG_USB_CONFIGFS_OBEX=NOT_SET
CONFIG_USB_CONFIGFS_NCM=y
CONFIG_USB_CONFIGFS_ECM=NOT_SET
CONFIG_USB_CONFIGFS_ECM_SUBSET=NOT_SET
CONFIG_USB_CONFIGFS_RNDIS=y
CONFIG_USB_CONFIGFS_EEM=NOT_SET
CONFIG_USB_CONFIGFS_PHONET=NOT_SET
CONFIG_USB_CONFIGFS_MASS_STORAGE=NOT_SET
CONFIG_USB_CONFIGFS_F_LB_SS=NOT_SET
CONFIG_USB_CONFIGFS_F_FS=y
CONFIG_USB_CONFIGFS_F_DM=y
CONFIG_USB_CONFIGFS_F_ADB=NOT_SET
CONFIG_USB_CONFIGFS_F_MTP=y
CONFIG_USB_CONFIGFS_F_PTP=y
CONFIG_USB_CONFIGFS_F_CONN_GADGET=y
CONFIG_USB_CONFIGFS_F_ACC=y
CONFIG_USB_CONFIGFS_F_AUDIO_SRC=y
CONFIG_USB_CONFIGFS_UEVENT=y
CONFIG_USB_CONFIGFS_F_UAC1=NOT_SET
CONFIG_USB_CONFIGFS_F_UAC2=NOT_SET
CONFIG_USB_CONFIGFS_F_MIDI=y
CONFIG_USB_CONFIGFS_F_HID=NOT_SET
CONFIG_USB_CONFIGFS_F_UVC=NOT_SET
CONFIG_USB_CONFIGFS_F_PRINTER=NOT_SET
CONFIG_IIO_CONFIGFS=NOT_SET
CONFIG_CONFIGFS_FS=y

ROOT (stderr) -

tejado, 2021-04-29:
> 
> 
> > 1. But using now the default system gadget (g1) does not let me really use ffs.hid or userspace functions. There are some hard dependencies to ADB which seems to cause conflicts with the underlying controller, e.g. hard disabling ffs.adb in the gadget removes the gadget<->controller binding (-> UDC empty). And using ffs.adb with ffs.hid together does not work (/dev/hidg* device is not created).
> 
> The DWC USB controller itself does not care, neither does driver in mainline kernel. The vendor could do many things with it, including disabling of specific feature etc.

Yes, I thought that the conflict is not inside the controller. But I want to understand it and fix it inside the source - if even possible.

tejado, 2021-04-29:
@CyberKid1987 good luck... I have an Samsung S8 here and so far I was not able to get it working due to the described issue of the ffs.adb conflict.

CyberKid1987, 2021-04-29:
ROOT - Execute command: for dir in /config/usb_gadget/*/; do echo GADGET_PATH=$dir; cd $dir/configs/; echo CONFIG_PATH="$dir/configs/`ls -1 | head -1`/"; cd $dir; if [ "$?" -ne "0" ]; then echo "Error - not able to change dir to $dir... exit"; exit 1; fi; echo UDC=$(cat UDC); find ./configs/ -type l -exec sh -c 'echo FUNCTIONS_ACTIVE=$(basename $(readlink "$@"))' _ {} \;; for f in ./functions/*/; do echo FUNCTIONS=$(basename $f); done; cd ./strings/0x409/; for vars in *; do echo ${vars}=$(cat $vars); done; echo "============="; done; 

ROOT (stdout) - GADGET_PATH=/config/usb_gadget/g1/
CONFIG_PATH=/config/usb_gadget/g1//configs/b.1/
UDC=10c00000.dwc3
FUNCTIONS_ACTIVE=ncm.0
FUNCTIONS_ACTIVE=audio_source.0
FUNCTIONS_ACTIVE=accessory.0
FUNCTIONS_ACTIVE=midi.0
FUNCTIONS_ACTIVE=conn_gadget.0
FUNCTIONS_ACTIVE=dm.0
FUNCTIONS_ACTIVE=rndis.0
FUNCTIONS_ACTIVE=acm.0
FUNCTIONS_ACTIVE=ffs.adb
FUNCTIONS_ACTIVE=ptp.0
FUNCTIONS_ACTIVE=mtp.0
FUNCTIONS=accessory.0
FUNCTIONS=acm.0
FUNCTIONS=acm.usb0
FUNCTIONS=audio_source.0
FUNCTIONS=conn_gadget.0
FUNCTIONS=dm.0
FUNCTIONS=ecm.usb0
FUNCTIONS=ffs.adb
FUNCTIONS=hid.0
FUNCTIONS=hid.1
FUNCTIONS=midi.0
FUNCTIONS=mtp.0
FUNCTIONS=ncm.0
FUNCTIONS=ptp.0
FUNCTIONS=rndis.0
manufacturer=SAMSUNG
product=SAMSUNG_Android
serialnumber=27c9f8a185217ece
=============

ROOT (stderr) - 
ROOT - Execute command: echo KERNEL_VERSION=`(uname -r |cut -d '-' -f1 )` && (gunzip -c /proc/config.gz | grep -i configfs | sed 's/# //; s/ is not set/=NOT_SET/')

ROOT (stdout) - KERNEL_VERSION=4.9.218
CONFIG_USB_CONFIGFS=y
CONFIG_USB_CONFIGFS_SERIAL=NOT_SET
CONFIG_USB_CONFIGFS_ACM=y
CONFIG_USB_CONFIGFS_OBEX=NOT_SET
CONFIG_USB_CONFIGFS_NCM=y
CONFIG_USB_CONFIGFS_ECM=NOT_SET
CONFIG_USB_CONFIGFS_ECM_SUBSET=NOT_SET
CONFIG_USB_CONFIGFS_RNDIS=y
CONFIG_USB_CONFIGFS_EEM=NOT_SET
CONFIG_USB_CONFIGFS_PHONET=NOT_SET
CONFIG_USB_CONFIGFS_MASS_STORAGE=NOT_SET
CONFIG_USB_CONFIGFS_F_LB_SS=NOT_SET
CONFIG_USB_CONFIGFS_F_FS=y
CONFIG_USB_CONFIGFS_F_DM=y
CONFIG_USB_CONFIGFS_F_ADB=NOT_SET
CONFIG_USB_CONFIGFS_F_MTP=y
CONFIG_USB_CONFIGFS_F_PTP=y
CONFIG_USB_CONFIGFS_F_CONN_GADGET=y
CONFIG_USB_CONFIGFS_F_ACC=y
CONFIG_USB_CONFIGFS_F_AUDIO_SRC=y
CONFIG_USB_CONFIGFS_UEVENT=y
CONFIG_USB_CONFIGFS_F_UAC1=NOT_SET
CONFIG_USB_CONFIGFS_F_UAC2=NOT_SET
CONFIG_USB_CONFIGFS_F_MIDI=y
CONFIG_USB_CONFIGFS_F_HID=NOT_SET
CONFIG_USB_CONFIGFS_F_UVC=NOT_SET
CONFIG_USB_CONFIGFS_F_PRINTER=NOT_SET
CONFIG_IIO_CONFIGFS=NOT_SET
CONFIG_CONFIGFS_FS=y

ROOT (stderr) - 

> @CyberKid1987 good luck... I have an Samsung S8 here and so far I was not able to get it working due to the described issue of the ffs.adb conflict.

On Note 2,3,4  works perfect. On 8,9,10 I cant enable.Only someone of Samsung kernel Developers can help us..

tejado, 2021-04-29:
> On Note 2,3,4 works perfect. On 8,9,10 I cant enable.Only someone of Samsung kernel Developers can help us..

On the Note it is working? This could be helpful. Do you have a kernel source from a working Note ROM? Then I can dig into it.

CyberKid1987, 2021-04-29:
ROOT - Execute command: for dir in /config/usb_gadget/*/; do echo GADGET_PATH=$dir; cd $dir/configs/; echo CONFIG_PATH="$dir/configs/`ls -1 | head -1`/"; cd $dir; if [ "$?" -ne "0" ]; then echo "Error - not able to change dir to $dir... exit"; exit 1; fi; echo UDC=$(cat UDC); find ./configs/ -type l -exec sh -c 'echo FUNCTIONS_ACTIVE=$(basename $(readlink "$@"))' _ {} \;; for f in ./functions/*/; do echo FUNCTIONS=$(basename $f); done; cd ./strings/0x409/; for vars in *; do echo ${vars}=$(cat $vars); done; echo "============="; done; 

ROOT (stdout) - GADGET_PATH=/config/usb_gadget/g1/
CONFIG_PATH=/config/usb_gadget/g1//configs/b.1/
UDC=10c00000.dwc3
FUNCTIONS_ACTIVE=ncm.0
FUNCTIONS_ACTIVE=audio_source.0
FUNCTIONS_ACTIVE=accessory.0
FUNCTIONS_ACTIVE=midi.0
FUNCTIONS_ACTIVE=conn_gadget.0
FUNCTIONS_ACTIVE=dm.0
FUNCTIONS_ACTIVE=rndis.0
FUNCTIONS_ACTIVE=acm.0
FUNCTIONS_ACTIVE=ffs.adb
FUNCTIONS_ACTIVE=ptp.0
FUNCTIONS_ACTIVE=mtp.0
FUNCTIONS=accessory.0
FUNCTIONS=acm.0
FUNCTIONS=acm.usb0
FUNCTIONS=audio_source.0
FUNCTIONS=conn_gadget.0
FUNCTIONS=dm.0
FUNCTIONS=ecm.usb0
FUNCTIONS=ffs.adb
FUNCTIONS=hid.0
FUNCTIONS=hid.1
FUNCTIONS=midi.0
FUNCTIONS=mtp.0
FUNCTIONS=ncm.0
FUNCTIONS=ptp.0
FUNCTIONS=rndis.0
manufacturer=SAMSUNG
product=SAMSUNG_Android
serialnumber=27c9f8a185217ece
=============

ROOT (stderr) - 
ROOT - Execute command: echo KERNEL_VERSION=`(uname -r |cut -d '-' -f1 )` && (gunzip -c /proc/config.gz | grep -i configfs | sed 's/# //; s/ is not set/=NOT_SET/')

ROOT (stdout) - KERNEL_VERSION=4.9.218
CONFIG_USB_CONFIGFS=y
CONFIG_USB_CONFIGFS_SERIAL=NOT_SET
CONFIG_USB_CONFIGFS_ACM=y
CONFIG_USB_CONFIGFS_OBEX=NOT_SET
CONFIG_USB_CONFIGFS_NCM=y
CONFIG_USB_CONFIGFS_ECM=NOT_SET
CONFIG_USB_CONFIGFS_ECM_SUBSET=NOT_SET
CONFIG_USB_CONFIGFS_RNDIS=y
CONFIG_USB_CONFIGFS_EEM=NOT_SET
CONFIG_USB_CONFIGFS_PHONET=NOT_SET
CONFIG_USB_CONFIGFS_MASS_STORAGE=NOT_SET
CONFIG_USB_CONFIGFS_F_LB_SS=NOT_SET
CONFIG_USB_CONFIGFS_F_FS=y
CONFIG_USB_CONFIGFS_F_DM=y
CONFIG_USB_CONFIGFS_F_ADB=NOT_SET
CONFIG_USB_CONFIGFS_F_MTP=y
CONFIG_USB_CONFIGFS_F_PTP=y
CONFIG_USB_CONFIGFS_F_CONN_GADGET=y
CONFIG_USB_CONFIGFS_F_ACC=y
CONFIG_USB_CONFIGFS_F_AUDIO_SRC=y
CONFIG_USB_CONFIGFS_UEVENT=y
CONFIG_USB_CONFIGFS_F_UAC1=NOT_SET
CONFIG_USB_CONFIGFS_F_UAC2=NOT_SET
CONFIG_USB_CONFIGFS_F_MIDI=y
CONFIG_USB_CONFIGFS_F_HID=NOT_SET
CONFIG_USB_CONFIGFS_F_UVC=NOT_SET
CONFIG_USB_CONFIGFS_F_PRINTER=NOT_SET
CONFIG_IIO_CONFIGFS=NOT_SET
CONFIG_CONFIGFS_FS=y

ROOT (stderr) - 

> @CyberKid1987 good luck... I have an Samsung S8 here and so far I was not able to get it working due to the described issue of the ffs.adb conflict.

On Note 2,3,4  works perfect. On 8,9,10 I cant enable.Only someone of Samsung kernel Developers can help us..

> > On Note 2,3,4 works perfect. On 8,9,10 I cant enable.Only someone of Samsung kernel Developers can help us..
> 
> On the Note it is working? This could be helpful. Do you have a kernel source from a working Note ROM? Then I can dig into it.

Yes but have 3.x kernel version.

tejado, 2021-04-29:
> > > On Note 2,3,4 works perfect. On 8,9,10 I cant enable.Only someone of Samsung kernel Developers can help us..
> > 
> > 
> > On the Note it is working? This could be helpful. Do you have a kernel source from a working Note ROM? Then I can dig into it.
> 
> Yes but have 3.x kernel version.

Please don't paste always the log as it will hard to follow the github issue. Thank you.
Ah okay, I thought the Note devices were more recent.

yesimxev, 2021-04-30:
@tejado finally had some time to pop in here. Note 9 has kernel 4.9.x. So where are we at:
1. Built a latest LOS17.1 kernel with features enabled - this time settings worked - CONFIGFS_F_x changes weren't effective with an older source
2. Functions could be enabled through your app, but weren't effective.
3. Tried our init.nethunter.rc and descriptors, no success - We have our own init file for initialising gadgets (some phones doesn't let NetHunter app to enable features without that)
4. Patched with generic-4.9 patch, without init files, still no success 
5. Patched with generic-4.9 patch, with init files, still no success
6. Patched tonight only part of generic-4.9 patch (from line 175 to 191 in the below patch)  to have the HID init in kernel, and functions could be still enabled without init files - like on Google Pixel 3a @CyberKid1987 will confirm tomorrow if success

https://github.com/pelya/android-keyboard-gadget/blob/master/patches/existing_tested/by-generic-kernel-version/generic_kernel_4.9.patch

I'll push our patches and defconfig, without HID patch
https://github.com/yesimxev/android_kernel_samsung_universal9810

This init version works for most phone (which needs it at all)
https://gitlab.com/kalilinux/nethunter/build-scripts/kali-nethunter-devices/-/tree/master/ten/oneplus7-oos/ramdisk

tejado, 2021-04-30:
Yes, disabling ADB needs to be done so HID function can work (as I wrote above).
Do you have dmesg messages of all your tries? So there might be an error message why certain steps failed.

CyberKid1987, 2021-04-30:
> > ```
> > 1. yesimxev fix it but not working hid 🙂
> > ```
> 
> Sorry, what do you mean? :-) Is the kernel config now showing that it is enabled?

Yes, show enable but usb gadget not working

CyberKid1987, 2021-04-30:
Dmesg file is here https://fromsmash.com/nfV.kUxWc0-dt

With New xevs kernel when i give su -c setprop sys.usb.config win,mtp,hid,adb;exit I see hid.0-1 on usb gadget. 
[image]

CyberKid1987, 2021-05-01:
New dmesg after reboot, type + mouse and keyboard

https://fromsmash.com/7H3_CeHGpQ-dt