tejado/android-usb-gadget #13: i grant root to app but when i open this app i cant show anything in Gadgets section

Zeel2323-coder, 2021-04-22:
this is my log
################################
ROOT - Execute command: for dir in /config/usb_gadget//; do echo GADGET_PATH=$dir; cd $dir/configs/; echo CONFIG_PATH="$dir/configs/ls -1 | head -1/"; cd $dir; if [ "$?" -ne "0" ]; then echo "Error - not able to change dir to $dir... exit"; exit 1; fi; echo UDC=$(cat UDC); find ./configs/ -type l -exec sh -c 'echo FUNCTIONS_ACTIVE=$(basename $(readlink "$@"))' _ {} ;; for f in ./functions//; do echo FUNCTIONS=$(basename $f); done; cd ./strings/0x409/; for vars in *; do echo ${vars}=$(cat $vars); done; echo "============="; done;

ROOT (stdout) - GADGET_PATH=/config/usb_gadget//
CONFIG_PATH=/config/usb_gadget///configs/acct/
Error - not able to change dir to /config/usb_gadget/*/... exit

ROOT (stderr) - : [1]: cd: /config/usb_gadget//configs: No such file or directory
: [1]: cd: /config/usb_gadget/: No such file or directory
root - length=1; index=1
ROOT - Execute command: echo KERNEL_VERSION=(uname -r |cut -d '-' -f1 ) && (gunzip -c /proc/config.gz | grep -i configfs | sed 's/# //; s/ is not set/=NOT_SET/')

ROOT (stdout) - KERNEL_VERSION=4.4.83+
CONFIG_USB_CONFIGFS=NOT_SET
CONFIG_CONFIGFS_FS=y

ROOT (stderr) -
##########################
[image]
[image]

tejado, 2021-04-29:
What device and ROM are you using? 
It seems that you have a more up to date kernel but it does not have ConfigFS enabked - this is pretty strange as ConfigFS is also used for ADB and other Android specific features.

Zeel2323-coder, 2021-04-29:
My device name is Lenovo k8 plus
This my phone's about section.
You can check everything.
If you still have any questions then ask me i reply that but please help me
to solve this problem.

Zeel2323-coder, 2021-04-29:
And ya one thing when I reinstall this app after install busybox app
utility it will send that error message in log file which you can see in
image which is uploaded on github issues. And when i install that without
busybox that time I get this error. In image. Check it out.

tejado, 2021-04-29:
Can you please execute following command and paste here the output (if any)?
`mount | grep -i config `

Zeel2323-coder, 2021-04-29:
At which place i can execute this command ??
Termux?

tejado, 2021-04-29:
Yes, or any other terminal app.

Zeel2323-coder, 2021-04-29:
Ok but in terminal where i can execute this command ?? In root directory ?
Please Tell specific directory.

Zeel2323-coder, 2021-04-29:
This is image after execute that command

tejado, 2021-04-29:
I'm not able to see your images... did you send it per mail? Please upload it directly.

Zeel2323-coder, 2021-04-29:
Yes, I did.
This picture is more zoomed and clear. Check out this image.

tejado, 2021-04-29:
As I said... I don't see any images here you send per mail...

Zeel2323-coder, 2021-04-29:
What why how can i send image?

Zeel2323-coder, 2021-04-29:
Can we talk in telegram. I will give my number to you.

Zeel2323-coder, 2021-04-29:
Can receive this image.?

tejado, 2021-04-29:
You can upload it here if you go with your web browser to this issue...

Zeel2323-coder, 2021-04-29:
[image]
[image]

Zeel2323-coder, 2021-04-29:
[image]

Zeel2323-coder, 2021-04-29:
[image]

Zeel2323-coder, 2021-04-29:
Now,You can check images.

tejado, 2021-04-29:
Great. Can you execute the following please - with the root account?
`ls -la /config/`

Zeel2323-coder, 2021-04-29:
Ohh bro at this time I can't because I from India and in India time is 12:30 I execute this command later and I show you the output tomorrow OK.  So sorry bro 😅

Zeel2323-coder, 2021-04-30:
[image]

Zeel2323-coder, 2021-04-30:
This is output of that command which is send by you check out this.

tejado, 2021-04-30:
Okay, strange. In your first message you have the error "cd: /config/usb_gadget/: No such file or directory" but you can see it on your screen. What is the content of usb_gadget (`ls -la /config/usb_gadget/`)

Zeel2323-coder, 2021-04-30:
[image]

Zeel2323-coder, 2021-04-30:
Check this

tejado, 2021-04-30:
Okay, can you add a new gadget over the app? Over the plus sign on the top right corner.

Zeel2323-coder, 2021-04-30:
I get nothing in new gadgets see this

Zeel2323-coder, 2021-04-30:
[image]

tejado, 2021-04-30:
Ohh, this seems to be a bug in my app :/ strange.
Do you see anything in logcat for this app when you are opening the Add gadget popup?

Zeel2323-coder, 2021-04-30:
No

Zeel2323-coder, 2021-04-30:
Wait I share frash log after this you can see

Zeel2323-coder, 2021-04-30:
[image]
In this image before gunzip which I marked that log is when I open app first time and after log is when I click to refresh button in top right side of corner near add gadget.

tejado, 2021-05-16:
This is interessting - seems there are several issues here but I'm not yet sure if they are related to each other.
I will do some tests on my own.

In the meantime - can you provide the log by logcat if you start the program? https://developer.android.com/studio/command-line/logcat

tejado, 2021-06-03:
@Zeel2323-coder any news from your side?

Zeel2323-coder, 2021-06-06:
Sorry for late my phone cant boot when i reboot it automatically boot in twrp recovery and screen is also blinking so I also cant use twrp. I think my smartphone is brick. If you know how to fix then please help. 😢

closed 2021-06-06 by tejado

Zeel2323-coder, 2021-06-06:
Hey bro i unbrick my phone using stoke rom flash now i can use my phone successfuly. Tell me what you want to ask me.

Zeel2323-coder, 2021-06-06:
Hey bro please reopen that issue i already unbrick my phone today. So we
can discuss more please 😀

tejado, 2021-06-06:
Yes, but please keep in mind, that this is not an support forum ;)
I already asked for a detailed logcat during the start of the app. If you can provide further details, I will reopen the issue.

Zeel2323-coder, 2021-06-06:
Ya ya sure why not?  I have a question for you in you github account there
is new 4.0 beta release  and previously I install  3.0 beta  release. Then
which apk I install ?? After I sent logs .