tejado/android-usb-gadget #14: Support SDK 25?

DRSDavidSoft, 2021-04-28:
It seems that the minimum supported SDK version is currently [26](https://github.com/tejado/android-usb-gadget/blob/master/app/build.gradle#L8).  Is it possible to also support SDK 25 (i.e., Android 7)? I have a LineageOS device for which only ROMs up to Android 7 are available, thus the installation fails:

```
Failed to install B:\app-release.apk: Failure [INSTALL_FAILED_OLDER_SDK: Failed parse during installPackageLI: /data/app/vmdl1785514132.tmp/base.apk (at Binary XML file line #7): Requires newer sdk version #26 (current version is #25)]
```

tejado, 2021-04-29:
What is the kernel version of your ROMs? There is a hard dependency on the kernel so in most cases it does not make sense to support lower SDKs.

goshil, 2021-04-29:
https://github.com/tejado/android-usb-gadget/issues/5#issuecomment-829245262

DRSDavidSoft, 2021-04-29:
@tejado I see, the Android SDK 25 on my LineageOS device has the kernel version 3.4.67, which is too old to support this.  May I suggest that instead of limiting the minimum supported Android SDK the app checks for the kernel available?

It is possible to flash a kernel with Linux 4.x on this device, which supports the same ROM.  That way, the kernel version is independent of the actual Android version (e.g. 7.1.2 in this case).

tejado, 2021-04-29:
@DRSDavidSoft 
Yes, please see https://github.com/tejado/android-usb-gadget/issues/2 for it. Feel free to create an PR for it :)
And in most ROM's, the SDK version is related to the kernel version. It was easier to use this as a first check then implementing something like mentioned in the issue https://github.com/tejado/android-usb-gadget/issues/

tejado, 2021-05-10:
I will close this as it is basically also a duplicate of https://github.com/tejado/android-usb-gadget/issues/5

closed 2021-05-10 by tejado