tejado/android-usb-gadget #16: hid.keyboard not showing up

xorartharth, 2021-05-03:
Hello again, i've got a new issue. Maybe I'm doing something wrong, but the hid.keyboard won't show up. It's also not available if I click on the first gadget. Add function -> Keyboard does nothing. My Kernels version is 4.9.148, at USB Device it says hisi-usb-otg. Thanks for help :)

yashveer55, 2021-06-10:
Sir i use pycharm and i want to use dictionary attack   in any phone but pycharm provides practice environment  not prectical environment  so how can i perform it. I am a student so I don't know how to do it    please sir gide me

daren-k, 2021-06-18:
Hi @tejado I'm having the same issue with the hid.keyboard not showing on a Samsung S8+

tejado, 2021-06-21:
Hi all, can you please paste your log on pastebin and provide the link here, so I can checkout what happend?

daren-k, 2021-06-21:
`ROOT - Execute command: for dir in /config/usb_gadget/*/; do echo GADGET_PATH=$dir; cd $dir/configs/; echo CONFIG_PATH="$dir/configs/`ls -1 | head -1`/"; cd $dir; if [ "$?" -ne "0" ]; then echo "Error - not able to change dir to $dir... exit"; exit 1; fi; echo UDC=$(cat UDC); find ./configs/ -type l -exec sh -c 'echo FUNCTIONS_ACTIVE=$(basename $(readlink "$@"))' _ {} \;; for f in ./functions/*/; do echo FUNCTIONS=$(basename $f); done; cd ./strings/0x409/; for vars in *; do echo ${vars}=$(cat $vars); done; echo "============="; done; 

ROOT (stdout) - GADGET_PATH=/config/usb_gadget/g1/
CONFIG_PATH=/config/usb_gadget/g1//configs/b.1/
UDC=a800000.dwc3
FUNCTIONS_ACTIVE=ffs.adb
FUNCTIONS_ACTIVE=acm.0
FUNCTIONS_ACTIVE=mtp.gs0
FUNCTIONS=accessory.gs2
FUNCTIONS=acm.0
FUNCTIONS=audio_source.gs3
FUNCTIONS=conn_gadget.0
FUNCTIONS=cser.dun.0
FUNCTIONS=cser.nmea.1
FUNCTIONS=diag.diag
FUNCTIONS=ffs.adb
FUNCTIONS=gsi.dpl
FUNCTIONS=gsi.rmnet
FUNCTIONS=gsi.rndis
FUNCTIONS=mass_storage.0
FUNCTIONS=midi.gs5
FUNCTIONS=mtp.gs0
FUNCTIONS=ncm.0
FUNCTIONS=ptp.gs1
FUNCTIONS=qdss.qdss
manufacturer=samsung
product=SM-G955U
serialnumber=98899a4147524e484e
=============

ROOT (stderr) - 
ROOT - Execute command: echo KERNEL_VERSION=`(uname -r |cut -d '-' -f1 )` && (gunzip -c /proc/config.gz | grep -i configfs | sed 's/# //; s/ is not set/=NOT_SET/')

ROOT (stdout) - KERNEL_VERSION=4.4.16

ROOT (stderr) - gunzip: /proc/config.gz: No such file or directory
`

tejado, 2021-06-21:
Thanks for the log but there is no information about adding hid keyboard function to your device. Can you please do this in the app and then paste your log into [pastebin](https://pastebin.com/) ?

### PLEASE don't paste it again here.... a lot of logs will make the issues here hard to read and I have to delete your comments.

daren-k, 2021-06-21:
[image]
The app do not allow me to a new function

tejado, 2021-06-22:
@daren-k thank you very much. I already saw this issue by a different person. I will make some code changes, provide you a new APK and it would be great if you can test it then again.

daren-k, 2021-06-22:
> @daren-k thank you very much. I already saw this issue by a different person. I will make some code changes, provide you a new APK and it would be great if you can test it then again.

Sure, let me know and I will test

closed 2021-10-31 by xorartharth