tejado/android-usb-gadget #18: My Preferences rests after every Reboot

name-is-cipher, 2021-05-16:
I added hid.keyboard to g1, after I reboot it is gone. I even tried with my g2, which has hid.keyboard and hid.mouse, it also disappears after reboot.

tejado, 2021-05-17:
Yes - there is no persistence yet. It would be great to get some help on this feature :)
Also it is not easy to implement it as you have to be careful not crashing your phone and so on....

name-is-cipher, 2021-05-18:
Found out why it is happening, boot IMG checks for tampering on every boot, And restores it to default system IMG if found any. The only way you can have persistence is to compile a custom kernel with hid patch enabled.  
Anyways tejado you have done great work bro !!!  
Trust me when I say, compiling kernel that to the stock kernel is not an easy task.

tejado, 2021-05-18:
No, this is not the only way :)
I plan to implement an startup action into the app which sets the specified gadgets at device boot. Android is doing it the same with the initrc usb scripts.

name-is-cipher, 2021-05-18:
Oh Damn, that's awesome !!!
A small suggestion, don't integrate it inside the app, give us a switch option whether to start on bootup or not. 
I particularly don't like app starting at bootup running in BG.
XD

tejado, 2022-10-29:
Added over #40

closed 2022-10-29 by tejado