tejado/android-usb-gadget #19: UVC for webcam

the-psychedelic, 2021-05-19:
Will there be an update to support uvc to allow the device as a pnp webcam?

yashpalgoyal1304, 2021-05-28:
[uvc](https://duckduckgo.com/?q=uvc&t=vivaldi&ia=web) = USB Video Class

i found this app via https://gitlab.com/fdroid/rfp/-/issues/1747 , and this pnp webcam was the first thing that came in my mind. thanks for opening this issue @the-psychedelic

tejado, 2021-05-31:
I can add UVC config for USB Gadget. But I guess there needs to be some additional app on the phone, so the device camera is forwarding it to the USB Gadget device?! Or do you have any information on how this can work?

the-psychedelic, 2021-05-31:
> I can add UVC config for USB Gadget. But I guess there needs to be some additional app on the phone, so the device camera is forwarding it to the USB Gadget device?! Or do you have any information on how this can work?

No I really have no information on how it would work. That is why I was hoping if you could help me. And what app should I use, then?

tejado, 2021-06-03:
I added the UVC camera function profile in the latest 0.4-beta version. Of course, your kernel needs to have this enabled (config_usb_configfs_f_uvc)

Unfortunately, I was not able to test it yet. For testing it, you will need some userspace app, e.g. you could try https://git.ideasonboard.org/uvc-gadget.git or https://github.com/jerome2016/Android_USB_UVC_Camera.

yashpalgoyal1304, 2021-06-03:
> I added the UVC camera function profile

 7f3cdfcff364f205f6a107f2a974033585225f1a

yashpalgoyal1304, 2022-02-13:
hi @tejado (sorry to ping you while busy)
can it be closed?

tejado, 2022-10-02:
Closing this issue. Current status and discussion in https://github.com/tejado/android-usb-gadget/issues/38

closed 2022-10-02 by tejado