tejado/android-usb-gadget #24: Question: Does this support using the android device itself as a USB camera?

tejaschaphalkar, 2021-07-22:
With the USB Gadget Tool, will I be able to connect my phone to a PC and use it as a camera?

boognish-rising, 2022-11-27:
I know this is over a year old but you never received an answer so while I can't answer your question directly, I can suggest 2 alternatives that work well

DroidCam

Iriun Webcam

The first is the better known of the two but also comes with ads unless you purchase the premium version

I personally prefer Iriun not just because it's free and without ads but also seems to just work better have a nicer interface and more features but your mileage may vary

tejado, 2022-11-28:
At the current standpoint, this project - android-usb-gadget - does not provide any implementation logic of USB gadgets. It only configures them, so that additional apps like Authorizer can use them.

We are currently working on a UVC camera integration, including a new feature in the app, to demonstrate the usage of USB gadgets.
Since DroidCam and Iriun Webcam require installing their own software on the respective system, it does not meet our requirements for such a solution.

Closing this issue. Current status and discussion in https://github.com/tejado/android-usb-gadget/issues/38

closed 2022-11-28 by tejado