tejado/android-usb-gadget #25: App is crashing..

Imman-coder, 2021-07-24:
07-24 21:39:24.404 7691-7691/? I/art: Late-enabling -Xcheck:jni
07-24 21:39:24.528 7691-7691/? W/System: ClassLoader referenced unknown path: /data/app/net.tjado.usbgadget-1/lib/arm
07-24 21:39:24.713 7691-7691/net.tjado.usbgadget W/art: Before Android 4.1, method android.graphics.PorterDuffColorFilter androidx.vectordrawable.graphics.drawable.VectorDrawableCompat.updateTintFilter(android.graphics.PorterDuffColorFilter, android.content.res.ColorStateList, android.graphics.PorterDuff$Mode) would have incorrectly overridden the package-private method in android.graphics.drawable.Drawable
07-24 21:39:24.821 7691-7691/net.tjado.usbgadget V/BoostFramework: mAcquireFunc method = public int com.qualcomm.qti.Performance.perfLockAcquire(int,int[])
07-24 21:39:24.821 7691-7691/net.tjado.usbgadget V/BoostFramework: mReleaseFunc method = public int com.qualcomm.qti.Performance.perfLockRelease()
07-24 21:39:24.821 7691-7691/net.tjado.usbgadget V/BoostFramework: mAcquireTouchFunc method = public int com.qualcomm.qti.Performance.perfLockAcquireTouch(android.view.MotionEvent,android.util.DisplayMetrics,int,int[])
07-24 21:39:24.821 7691-7691/net.tjado.usbgadget V/BoostFramework: mIOPStart method = public int com.qualcomm.qti.Performance.perfIOPrefetchStart(int,java.lang.String)
07-24 21:39:24.821 7691-7691/net.tjado.usbgadget V/BoostFramework: mIOPStop method = public int com.qualcomm.qti.Performance.perfIOPrefetchStop()
07-24 21:39:24.827 7691-7691/net.tjado.usbgadget V/BoostFramework: BoostFramework() : mPerf = com.qualcomm.qti.Performance@c02fdb3
07-24 21:39:24.828 7691-7691/net.tjado.usbgadget V/BoostFramework: BoostFramework() : mPerf = com.qualcomm.qti.Performance@204bf70
07-24 21:39:24.842 7691-7691/net.tjado.usbgadget I/art: Rejecting re-init on previously-failed class java.lang.Class<androidx.core.view.ViewCompat$2>
07-24 21:39:24.844 7691-7691/net.tjado.usbgadget I/art: Rejecting re-init on previously-failed class java.lang.Class<androidx.core.view.ViewCompat$2>
07-24 21:39:25.090 7691-7691/net.tjado.usbgadget V/BoostFramework: BoostFramework() : mPerf = com.qualcomm.qti.Performance@b74d35d
07-24 21:39:25.090 7691-7691/net.tjado.usbgadget V/BoostFramework: BoostFramework() : mPerf = com.qualcomm.qti.Performance@1284d2
07-24 21:39:25.213 7691-7691/net.tjado.usbgadget V/BoostFramework: BoostFramework() : mPerf = com.qualcomm.qti.Performance@701c1e
07-24 21:39:25.214 7691-7691/net.tjado.usbgadget V/BoostFramework: BoostFramework() : mPerf = com.qualcomm.qti.Performance@5814cff
07-24 21:39:25.222 7691-7691/net.tjado.usbgadget V/BoostFramework: BoostFramework() : mPerf = com.qualcomm.qti.Performance@9bdaacc
07-24 21:39:25.484 7691-7764/net.tjado.usbgadget D/OpenGLRenderer: Use EGL_SWAP_BEHAVIOR_PRESERVED: true
07-24 21:39:25.522 7691-7756/net.tjado.usbgadget D/ROOT: Execute command: for dir in /config/usb_gadget/*/; do echo GADGET_PATH=$dir; cd $dir/configs/; echo CONFIG_PATH="$dir/configs/`ls -1 | head -1`/"; cd $dir; if [ "$?" -ne "0" ]; then echo "Error - not able to change dir to $dir... exit"; exit 1; fi; echo UDC=$(cat UDC); find ./configs/ -type l -exec sh -c 'echo FUNCTIONS_ACTIVE=$(basename $(readlink "$@"))' _ {} \;; for f in ./functions/*/; do echo FUNCTIONS=$(basename $f); done; cd ./strings/0x409/; for vars in *; do echo ${vars}=$(cat $vars); done; echo "============="; done; 
07-24 21:39:25.586 7691-7764/net.tjado.usbgadget I/Adreno-EGL: <qeglDrvAPI_eglInitialize:379>: EGL 1.4 QUALCOMM build: Nondeterministic_AU_msm8909_LA.BR.1.2.7.C1__release_AU (I741a3d36ca)
    OpenGL ES Shader Compiler Version: E031.29.00.00
    Build Date: 09/22/16 Thu
    Local Branch: 
    Remote Branch: quic/LA.BR.1.2.7.c1
    Local Patches: NONE
    Reconstruct Branch: NOTHING
07-24 21:39:25.590 7691-7764/net.tjado.usbgadget I/OpenGLRenderer: Initialized EGL, version 1.4
07-24 21:39:25.699 7691-7691/net.tjado.usbgadget V/BoostFramework: BoostFramework() : mPerf = com.qualcomm.qti.Performance@9323824
07-24 21:39:25.700 7691-7691/net.tjado.usbgadget V/BoostFramework: BoostFramework() : mPerf = com.qualcomm.qti.Performance@7550f8d
07-24 21:39:25.753 7691-7691/net.tjado.usbgadget W/ResourceType: Failure getting entry for 0x7f070062 (t=6 e=98) (error -75)
07-24 21:39:25.763 7691-7691/net.tjado.usbgadget W/ResourceType: Failure getting entry for 0x7f07005f (t=6 e=95) (error -75)
07-24 21:39:25.763 7691-7691/net.tjado.usbgadget W/ResourceType: Failure getting entry for 0x7f070061 (t=6 e=97) (error -75)
07-24 21:39:25.784 7691-7691/net.tjado.usbgadget W/ResourceType: Failure getting entry for 0x7f070062 (t=6 e=98) (error -75)
07-24 21:39:25.785 7691-7691/net.tjado.usbgadget D/AndroidRuntime: Shutting down VM
    
    
    --------- beginning of crash
07-24 21:39:25.787 7691-7691/net.tjado.usbgadget E/AndroidRuntime: FATAL EXCEPTION: main
    Process: net.tjado.usbgadget, PID: 7691
    android.content.res.Resources$NotFoundException: Resource ID #0x7f070062
        at android.content.res.Resources.getValue(Resources.java:1351)
        at androidx.appcompat.widget.ResourceManagerInternal.loadDrawableFromDelegates(ResourceManagerInternal.java:252)
        at androidx.appcompat.widget.ResourceManagerInternal.getDrawable(ResourceManagerInternal.java:139)
        at androidx.appcompat.widget.ResourceManagerInternal.getDrawable(ResourceManagerInternal.java:132)
        at androidx.appcompat.content.res.AppCompatResources.getDrawable(AppCompatResources.java:104)
        at androidx.appcompat.view.menu.MenuItemImpl.getIcon(MenuItemImpl.java:505)
        at androidx.appcompat.view.menu.ActionMenuItemView.initialize(ActionMenuItemView.java:126)
        at androidx.appcompat.widget.ActionMenuPresenter.bindItemView(ActionMenuPresenter.java:207)
        at androidx.appcompat.view.menu.BaseMenuPresenter.getItemView(BaseMenuPresenter.java:188)
        at androidx.appcompat.widget.ActionMenuPresenter.getItemView(ActionMenuPresenter.java:193)
        at androidx.appcompat.widget.ActionMenuPresenter.flagActionItems(ActionMenuPresenter.java:473)
        at androidx.appcompat.view.menu.MenuBuilder.flagActionItems(MenuBuilder.java:1182)
        at androidx.appcompat.view.menu.BaseMenuPresenter.updateMenuView(BaseMenuPresenter.java:96)
        at androidx.appcompat.widget.ActionMenuPresenter.updateMenuView(ActionMenuPresenter.java:226)
        at androidx.appcompat.view.menu.MenuBuilder.dispatchPresenterUpdate(MenuBuilder.java:298)
        at androidx.appcompat.view.menu.MenuBuilder.onItemsChanged(MenuBuilder.java:1069)
        at androidx.appcompat.view.menu.MenuBuilder.startDispatchingItemsChanged(MenuBuilder.java:1096)
        at androidx.appcompat.app.AppCompatDelegateImpl.preparePanel(AppCompatDelegateImpl.java:1788)
        at androidx.appcompat.app.AppCompatDelegateImpl.doInvalidatePanelMenu(AppCompatDelegateImpl.java:2026)
        at androidx.appcompat.app.AppCompatDelegateImpl$2.run(AppCompatDelegateImpl.java:263)
        at android.os.Handler.handleCallback(Handler.java:739)
        at android.os.Handler.dispatchMessage(Handler.java:95)
        at android.os.Looper.loop(Looper.java:148)
        at android.app.ActivityThread.main(ActivityThread.java:5417)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:726)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:616)
        at de.robv.android.xposed.XposedBridge.main(XposedBridge.java:107)
07-24 21:39:25.936 7691-7756/net.tjado.usbgadget D/ROOT (stdout): GADGET_PATH=/config/usb_gadget/*/
    CONFIG_PATH=/config/usb_gadget/*//configs//
    Error - not able to change dir to /config/usb_gadget/*/... exit
07-24 21:39:25.936 7691-7756/net.tjado.usbgadget D/ROOT (stderr): : <stdin>[1]: cd: /config/usb_gadget/*/configs: No such file or directory
    ls: Unknown option '-1'. Aborting.
    : <stdin>[1]: cd: /config/usb_gadget/*: No such file or directory

worstperson, 2022-02-06:
It looks like your phone doesn't have the usb_gadget module. I think the minsdk for this project is set to 23, but it almost certainly will not run pre-Oreo.
https://source.android.com/devices/tech/config/usb-hal

tejado, 2022-10-03:
In case this is still relevant, please open a new issue and provide details like Android version, ... 
Thank you!

closed 2022-10-03 by tejado