tejado/android-usb-gadget #27: no gadgets

Leox-06, 2021-08-31:
this is the logs:
```
ROOT - Execute command: for dir in /config/usb_gadget/*/; do echo GADGET_PATH=$dir; cd $dir/configs/; echo CONFIG_PATH="$dir/configs/ls -1 | head -1/"; cd $dir; if [ "$?" -ne "0" ]; then echo "Error - not able to change dir to $dir... exit"; exit 1; fi; echo UDC=$(cat UDC); find ./configs/ -type l -exec sh -c 'echo FUNCTIONS_ACTIVE=$(basename $(readlink "$@"))' _ {} \;; for f in ./functions/*/; do echo FUNCTIONS=$(basename $f); done; cd ./strings/0x409/; for vars in *; do echo ${vars}=$(cat $vars); done; echo "============="; done; 

ROOT (stdout) - GADGET_PATH=/config/usb_gadget/*/
CONFIG_PATH=/config/usb_gadget/*//configs/acct/
Error - not able to change dir to /config/usb_gadget/*/... exit

ROOT (stderr) - : <stdin>[1]: cd: /config/usb_gadget/*/configs: No such file or directory
: <stdin>[1]: cd: /config/usb_gadget/*: No such file or directory
root - length=1; index=1
ROOT - Execute command: echo KERNEL_VERSION=(uname -r |cut -d '-' -f1 ) && (gunzip -c /proc/config.gz | grep -i configfs | sed 's/# //; s/ is not set/=NOT_SET/')

ROOT (stdout) - KERNEL_VERSION=4.4.83+

ROOT (stderr) - : <stdin>[1]: gunzip: not found
```

closed 2021-09-19 by Leox-06

logm1lo, 2022-07-05:
> this is the logs:
>
> ```
> ROOT - Execute command: for dir in /config/usb_gadget/*/; do echo GADGET_PATH=$dir; cd $dir/configs/; echo CONFIG_PATH="$dir/configs/ls -1 | head -1/"; cd $dir; if [ "$?" -ne [...]

How did you fix this ?

tejado, 2022-07-09:
@logm1lo 
Install gunzip or a busybox version with gunzip.