tejado/android-usb-gadget #28: Dont work

ArastaD0t, 2021-09-18:
ROOT - Execute command: for dir in /config/usb_gadget/*/; do echo GADGET_PATH=$dir; cd $dir/configs/; echo CONFIG_PATH="$dir/configs/`ls -1 | head -1`/"; cd $dir; if [ "$?" -ne "0" ]; then echo "Error - not able to change dir to $dir... exit"; exit 1; fi; echo UDC=$(cat UDC); find ./configs/ -type l -exec sh -c 'echo FUNCTIONS_ACTIVE=$(basename $(readlink "$@"))' _ {} \;; for f in ./functions/*/; do echo FUNCTIONS=$(basename $f); done; cd ./strings/0x409/; for vars in *; do echo ${vars}=$(cat $vars); done; echo "============="; done; 

ROOT (stdout) - GADGET_PATH=/config/usb_gadget/*/
CONFIG_PATH=/config/usb_gadget/*//configs/acct/
Error - not able to change dir to /config/usb_gadget/*/... exit

ROOT (stderr) - : <stdin>[1]: cd: /config/usb_gadget/*/configs: No such file or directory
: <stdin>[1]: cd: /config/usb_gadget/*: No such file or directory
root - length=1; index=1
ROOT - Execute command: echo KERNEL_VERSION=`(uname -r |cut -d '-' -f1 )` && (gunzip -c /proc/config.gz | grep -i configfs | sed 's/# //; s/ is not set/=NOT_SET/')

ROOT (stdout) - KERNEL_VERSION=3.4.113

ROOT (stderr) - gunzip: /proc/config.gz: No such file or directory
ROOT - Execute command: for dir in /config/usb_gadget/*/; do echo GADGET_PATH=$dir; cd $dir/configs/; echo CONFIG_PATH="$dir/configs/`ls -1 | head -1`/"; cd $dir; if [ "$?" -ne "0" ]; then echo "Error - not able to change dir to $dir... exit"; exit 1; fi; echo UDC=$(cat UDC); find ./configs/ -type l -exec sh -c 'echo FUNCTIONS_ACTIVE=$(basename $(readlink "$@"))' _ {} \;; for f in ./functions/*/; do echo FUNCTIONS=$(basename $f); done; cd ./strings/0x409/; for vars in *; do echo ${vars}=$(cat $vars); done; echo "============="; done; 

ROOT (stdout) - GADGET_PATH=/config/usb_gadget/*/
CONFIG_PATH=/config/usb_gadget/*//configs/acct/
Error - not able to change dir to /config/usb_gadget/*/... exit

ROOT (stderr) - : <stdin>[1]: cd: /config/usb_gadget/*/configs: No such file or directory
: <stdin>[1]: cd: /config/usb_gadget/*: No such file or directory
root - length=1; index=1
ROOT - Execute command: #!/bin/sh

CONFIGFS_DIR="/config"
GADGETS_PATH="${CONFIGFS_DIR}/usb_gadget"

GADGET="keyboard"
GADGET_PATH=${GADGETS_PATH}/${GADGET}

CONFIG_PATH="$GADGET_PATH/configs/c.1/"
STRINGS_PATH="$GADGET_PATH/strings/0x409/"

mkdir -p $CONFIG_PATH
mkdir -p $STRINGS_PATH

mkdir -p $GADGET_PATH/functions/hid.keyboard
cd $GADGET_PATH/functions/hid.keyboard

# HID protocol (according to USB spec: 1 for keyboard)
echo 1 > protocol
# device subclass
echo 1 > subclass
# number of bytes per record
echo 8 > report_length

# writing report descriptor
echo -ne \\x05\\x01\\x09\\x06\\xa1\\x01\\x05\\x07\\x19\\xe0\\x29\\xe7\\x15\\x00\\x25\\x01\\x75\\x01\\x95\\x08\\x81\\x02\\x95\\x01\\x75\\x08\\x81\\x03\\x95\\x05\\x75\\x01\\x05\\x08\\x19\\x01\\x29\\x05\\x91\\x02\\x95\\x01\\x75\\x03\\x91\\x03\\x95\\x06\\x75\\x08\\x15\\x00\\x25\\x65\\x05\\x07\\x19\\x00\\x29\\x65\\x81\\x00\\xc0 > report_desc

mkdir -p $GADGET_PATH/functions/hid.mouse
cd $GADGET_PATH/functions/hid.mouse

# HID protocol (according to USB spec: 2 for mouse)
echo 2 > protocol
# device subclass
echo 1 > subclass
# number of bytes per record
echo 4 > report_length

# writing report descriptor
echo -ne \\x05\\x01\\x09\\x02\\xa1\\x01\\x09\\x01\\xa1\\x00\\x05\\x09\\x19\\x01\\x29\\x05\\x15\\x00\\x25\\x01\\x95\\x05\\x75\\x01\\x81\\x02\\x95\\x01\\x75\\x03\\x81\\x01\\x05\\x01\\x09\\x30\\x09\\x31\\x09\\x38\\x15\\x81\\x25\\x7F\\x75\\x08\\x95\\x03\\x81\\x06\\xc0\\xc0 > report_desc

cd $GADGET_PATH
echo 0x046a > idVendor
echo 0x002a > idProduct

cd $STRINGS_PATH
echo "tejado" > manufacturer
echo "Android USB Gadget" > product
echo "42" > serialnumber

cd $CONFIG_PATH
echo 120 > MaxPower
echo "Configuration" > strings/0x409/configuration

ln -s ${GADGET_PATH}/functions/hid.keyboard $CONFIG_PATH/hid.keyboard
ln -s ${GADGET_PATH}/functions/hid.mouse $CONFIG_PATH/hid.mouse

ROOT (stdout) - 
ROOT (stderr) - mkdir: '/config/usb_gadget': Operation not permitted
mkdir: '/config/usb_gadget': Operation not permitted
mkdir: '/config/usb_gadget': Operation not permitted
: <stdin>[16]: cd: /config/usb_gadget/keyboard/functions/hid.keyboard: No such file or directory
: <stdin>[19]: can't create protocol: Read-only file system
: <stdin>[21]: can't create subclass: Read-only file system
: <stdin>[23]: can't create report_length: Read-only file system
: <stdin>[26]: can't create report_desc: Read-only file system
mkdir: '/config/usb_gadget': Operation not permitted
: <stdin>[29]: cd: /config/usb_gadget/keyboard/functions/hid.mouse: No such file or directory
: <stdin>[32]: can't create protocol: Read-only file system
: <stdin>[34]: can't create subclass: Read-only file system
: <stdin>[36]: can't create report_length: Read-only file system
: <stdin>[39]: can't create report_desc: Read-only file system
: <stdin>[42]: cd: /config/usb_gadget/keyboard: No such file or directory
: <stdin>[43]: can't create idVendor: Read-only file system
: <stdin>[44]: can't create idProduct: Read-only file system
: <stdin>[46]: cd: /config/usb_gadget/keyboard/strings/0x409: No such file or directory
: <stdin>[47]: can't create manufacturer: Read-only file system
: <stdin>[48]: can't create product: Is a directory
: <stdin>[49]: can't create serialnumber: Read-only file system
: <stdin>[51]: cd: /config/usb_gadget/keyboard/configs/c.1: No such file or directory
: <stdin>[52]: can't create MaxPower: Read-only file system
: <stdin>[53]: can't create strings/0x409/configuration: No such file or directory
ln: cannot create symbolic link from '/config/usb_gadget/keyboard/functions/hid.keyboard' to '/config/usb_gadget/keyboard/configs/c.1//hid.keyboard': No such file or directory
ln: cannot create symbolic link from '/config/usb_gadget/keyboard/functions/hid.mouse' to '/config/usb_gadget/keyboard/configs/c.1//hid.mouse': No such file or directory
ROOT - Execute command: for dir in /config/usb_gadget/*/; do echo GADGET_PATH=$dir; cd $dir/configs/; echo CONFIG_PATH="$dir/configs/`ls -1 | head -1`/"; cd $dir; if [ "$?" -ne "0" ]; then echo "Error - not able to change dir to $dir... exit"; exit 1; fi; echo UDC=$(cat UDC); find ./configs/ -type l -exec sh -c 'echo FUNCTIONS_ACTIVE=$(basename $(readlink "$@"))' _ {} \;; for f in ./functions/*/; do echo FUNCTIONS=$(basename $f); done; cd ./strings/0x409/; for vars in *; do echo ${vars}=$(cat $vars); done; echo "============="; done; 

ROOT (stdout) - GADGET_PATH=/config/usb_gadget/*/
CONFIG_PATH=/config/usb_gadget/*//configs/acct/
Error - not able to change dir to /config/usb_gadget/*/... exit

ROOT (stderr) - : <stdin>[1]: cd: /config/usb_gadget/*/configs: No such file or directory
: <stdin>[1]: cd: /config/usb_gadget/*: No such file or directory
root - length=1; index=1

Kosaci, 2021-09-21:
Mount the configs as rw

tejado, 2022-10-03:
Kernel is too old.

closed 2022-10-03 by tejado