tejado/android-usb-gadget #31: Crash report

shuvashish76, 2022-01-20:
App crashes while click on either DEVICE INFO or LOGS

```
FATAL EXCEPTION: main
Process: net.tjado.usbgadget, PID: 6614
java.lang.NullPointerException: Attempt to invoke virtual method 'int java.lang.String.length()' on a null object reference
	at java.io.StringReader.<init>(StringReader.java:50)
	at net.tjado.usbgadget.GadgetShellApi.lambda$updateDeviceInfo$0(GadgetShellApi.java:52)
	at net.tjado.usbgadget.-$$Lambda$GadgetShellApi$2MbiYVtUUtsiThLGWXZp__L4lqc.OnRootTaskFinish(Unknown Source:2)
	at net.tjado.usbgadget.RootTask.onPostExecute(RootTask.java:37)
	at net.tjado.usbgadget.RootTask.onPostExecute(RootTask.java:6)
	at android.os.AsyncTask.finish(AsyncTask.java:695)
	at android.os.AsyncTask.-wrap1(Unknown Source:0)
	at android.os.AsyncTask$InternalHandler.handleMessage(AsyncTask.java:712)
	at android.os.Handler.dispatchMessage(Handler.java:106)
	at android.os.Looper.loop(Looper.java:164)
	at android.app.ActivityThread.main(ActivityThread.java:6518)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:438)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:807)
```

tejado, 2022-01-21:
Hi,

thanks for the issue. I will check it and come back to you!

worstperson, 2022-02-06:
I think some kernels in the wild are built without CONFIG_IKCONFIG_PROC enabled, I'm not sure it's a good idea to rely on it.

tejado, 2022-10-03:
This is a bug which normally occurs when the app has no access to root.
Fixed in https://github.com/tejado/android-usb-gadget/commit/7114806a163253bb6e0b7d9b06b627f9d5a07789 (not yet released).

closed 2022-10-03 by tejado