tejado/android-usb-gadget #35: Persistent functions?

Alexander-Shukaev, 2022-08-07:
Hi, I've noticed that rebooting a device results in all the additional functions configured to be reset.  Didn't find any setting for persisting the configuration so far.  Can this be a feature request or are there some Android limitations that prevent implementing this?

tejado, 2022-08-07:
Hi,
yes, this is on the roadmap: Creating & enabling USB Gadgets during boot.
It is possible but it will need to be carefully implemented as this could lead to an unbootable rom (then a reflash would be required).
So my plan is to implement a check, if the boot with certain settings was successful. If yes, it will be then persistet.

I cant say when this will be implemented but any help on this is highly appreciated.

tejado, 2022-10-03:
Duplication of https://github.com/tejado/android-usb-gadget/issues/18

closed 2022-10-03 by tejado

tejado, 2022-10-29:
Implemented by #40