tejado/android-usb-gadget #36: Not working on Samsung S10 5G HID gadget

himekifee, 2022-08-15:
I've used this lovely tool on another more recent Samsung device and that did work. However, on this relatively old device, it does not work due to #12 at first. It did not create the HID gadget profile as that out of memory popped out. Fortunately, following your discussion on how later android kernel source fixed that issue also works for me. I added `static int gadget_index` in the beginning and changed `device = device_create(android_class, NULL, MKDEV(0, 0), NULL, "android0");` to `device = device_create(android_class, NULL, MKDEV(0, 0), NULL, "android%d", gadget_index++);`. After that, I'm able to create a gadget as shown.
[image]

However, when I click on that enable button, it goes back to the disabled state automatically within 2 secs. I didn't really find useful info in logs.
[image]
[image]
May you give me some advice, please? I kinda need a solution as my machine is still down, and I don't have a keyboard...

tejado, 2022-08-16:
Hi @himekifee,
my personal solution is to avoid Samsung devices for that purpose. I'm currently looking at more "open" devices like Pixel.

In regard to further information: have a look into the system log (run logcat under root) during the time of enabling the gadget manually and auto-disabling. We might see there something related to it which helps to debug that. I tried to understand that behaviour on an old Samsung phone few years ago but never came to any conclusion as it was very very time consuming and my lack of knowledge about the gadget system in the kernel.

himekifee, 2022-08-16:
I was also able to add the `hid.keyboard` function in `g0` gadget. Is there a way to utilise this since I only do that for now as an emergent situation?

himekifee, 2022-08-16:
From what I can see there is no 0x409 dir in strings
```
:/config/usb_gadget/keyboard/configs/c.1/strings # ls
:/config/usb_gadget/keyboard/configs/c.1/strings # 
```
There is one under normal gadget
```
:/config/usb_gadget/g1/configs/b.1/strings # ls
0x409
```
Could it be the kernel thing that is fixed later as the above one?

himekifee, 2022-08-16:
I'm able to progress till I write `UDC`
```
[ 4693.039760] Config c/1 of keyboard needs at least one function.
[ 4693.039798] configfs-gadget 10c00000.dwc3: failed to start keyboard: -22
```
However, there are dirs under `configs/c.1`
```
130|:/config/usb_gadget/keyboard/configs/c.1 # ls
MaxPower  bmAttributes  hid.keyboard  hid.mouse  strings
```

tejado, 2022-08-16:
> I'm able to progress till I write `UDC`
>
> ```
> [ 4693.039760] Config c/1 of keyboard needs at least one function.
> [ 4693.039798] configfs-gadget 10c00000.dwc3: failed to start keyboard: -22
> ``` [...]

Here is the line of code which is reporting this error: https://github.com/corsicanu/android_kernel_samsung_universal9820/blob/7c38bfbe048f6b894b3f68d9bc7495399f0b0c87/drivers/usb/gadget/udc/core.c#L1353

MrRob0-X, 2023-02-19:
> This happens with exynos versions with 4.x kernels and up, as far as i know the cause of this problem is a device blobs, which prevents the creation of hidg* functions in the system. to solve this problem we need to create a config with usb controller model used on current device, we can use it as magisk module or so but sadly i don't know the method, i can only get the init file from device tree, This information was taken from a friend of mine @Tkpointz.

MrRob0-X, 2023-02-19:
> Hi @himekifee, my personal solution is to avoid Samsung devices for that purpose. I'm currently looking at more "open" devices like Pixel.
>
> In regard to further information: have a look into the [...]

This happens with exynos versions with 4.x kernels and up, as far as i know the cause of this problem is a device blobs, which prevents the creation of hidg* functions in the system. to solve this problem we need to create a config with usb controller model used on current device, we can use it as magisk module or so but sadly i don't know the method, i can only get the init file from device tree, This information was taken from a friend of mine @Tkpointz.

tejado, 2023-02-19:
> to solve this problem we need to **create a config with usb controller model used on current device**

What does this mean?

MrRob0-X, 2023-02-19:
> > to solve this problem we need to **create a config with usb controller model used on current device**
> 
> What does this mean?

a config like the normal init config

tejado, 2023-02-19:
A config or a gadget? So you are saying it needs to like the default one (g1)? And if it is a config, should it be in a separate gadget or in the default one?

MrRob0-X, 2023-02-19:
> A config or a gadget? So you are saying it needs to like the default one (g1)? And if it is a config, should it be in a separate gadget or in the default one?

Nah i meant init config like nethunter.blabla

tejado, 2023-02-19:
> > A config or a gadget? So you are saying it needs to like the default one (g1)? And if it is a config, should it be in a separate gadget or in the default one?
> 
> Nah i meant init config like nethunter.blabla

Sorry, dont understand what you mean. The USB init config (init.usb.rc) which basically configures the USB Gadget according to the system settings writes to configFS like android-usb-gadget does.

MrRob0-X, 2023-02-19:
> Sorry, dont understand what you mean. The USB init config (init.usb.rc) which basically configures the USB Gadget according to the system settings writes to configFS like android-usb-gadget does.

You are absolutely right, but the difference here is that modified config will be based on device init config like samsung a30s 
 as example ([init.exynos7885.usb.rc](https://github.com/eurekadevelopment/android_device_samsung_exynos7885/blob/android-13/universal7885-common/rootdir/etc/init.exynos7885.usb.rc))

tejado, 2023-02-19:
Thanks, I will try it - but currently, it is no priority and I general recommend Pixel phones for this.
Happy to see some other users implementing/testing this.

MrRob0-X, 2023-02-19:
> Thanks, I will try it - but currently, it is no priority and I general recommend Pixel phones for this. Happy to see some other users implementing/testing this.

Actually this is not a solution because i said earlier that i don't know how to create a custom config based on stock one, just got info about it.

About this problem, the only solution right now is to get an Exynos device with 3.x kernel version or snapragon version which works without problems, i have two samsung devices with snap chipset a70 and note 20 ultra and hid working like a charm