tejado/android-usb-gadget #37: Can you add usbserial support?

kk580kk, 2022-09-14:
Android device that not support USB serial, so I need a app can create ttyUSB0 file for python app read.

tejado, 2022-11-08:
I added a serial gadget using ACM function:
https://github.com/tejado/android-usb-gadget/commit/7e29b44dd5b9b12b3f84717ee40182d3b853493f

It might require a specific Android ROM which has ACM activated. I will look into other functions which could be an alternative to ACM abd are usually part of Android.

kk580kk, 2022-11-10:
My device is XIAOMI pad 5, and this rom didn't have usb serial driver lib.
```
1|nabu:/system/lib # modprobe -d /vendor/lib/modules usbserial vendor=0x1a86 product=0x7523
modprobe: alias lines in modules.alias must have 3 entries, not 5
modprobe: alias lines in modules.alias must have 3 entries, not 4
modprobe: alias lines in modules.alias must have 3 entries, not 4
modprobe: alias lines in modules.alias must have 3 entries, not 4
modprobe: LoadWithAliases was unable to load usbserial
modprobe: Failed to load module usbserial: No such file or directory
```

Can you add gadget serial support like 
https://android.googlesource.com/kernel/common/+/bcmdhd-3.10/Documentation/usb/gadget_serial.txt

eadmaster, 2024-06-02:
Is there an example on how to use this?
I simply like to echo all incoming messages from serial onscreen.