tejado/android-usb-gadget #42: Trying to write in an none existing folder ?

SF73, 2023-01-12:
Hi,
I'm currently trying to get this work on a Pixel 6A running android 13.

Reading the logs I noticed an issue:

So this is the interesting part of the logs

```sh
ROOT - Execute command: #!/bin/sh
 
CONFIGFS_DIR="/config"
GADGETS_PATH="${CONFIGFS_DIR}/usb_gadget"
 
GADGET="keyboard"
GADGET_PATH=${GADGETS_PATH}/${GADGET}
 
CONFIG_PATH="$GADGET_PATH/configs/c.1/"
STRINGS_PATH="$GADGET_PATH/strings/0x409/"
 
mkdir -p $CONFIG_PATH
mkdir -p $STRINGS_PATH
 
mkdir -p $GADGET_PATH/functions/hid.keyboard
cd $GADGET_PATH/functions/hid.keyboard
 
# HID protocol (according to USB spec: 1 for keyboard)
echo 1 > protocol
# device subclass
echo 1 > subclass
# number of bytes per record
echo 8 > report_length
 
# writing report descriptor
echo -ne \\x05\\x01\\x09\\x06\\xa1\\x01\\x05\\x07\\x19\\xe0\\x29\\xe7\\x15\\x00\\x25\\x01\\x75\\x01\\x95\\x08\\x81\\x02\\x95\\x01\\x75\\x08\\x81\\x03\\x95\\x05\\x75\\x01\\x05\\x08\\x19\\x01\\x29\\x05\\x91\\x02\\x95\\x01\\x75\\x03\\x91\\x03\\x95\\x06\\x75\\x08\\x15\\x00\\x25\\x65\\x05\\x07\\x19\\x00\\x29\\x65\\x81\\x00\\xc0 > report_desc

mkdir -p $GADGET_PATH/functions/hid.mouse
cd $GADGET_PATH/functions/hid.mouse
 
# HID protocol (according to USB spec: 2 for mouse)
echo 2 > protocol
# device subclass
echo 1 > subclass
# number of bytes per record
echo 4 > report_length
 
# writing report descriptor
echo -ne \\x05\\x01\\x09\\x02\\xa1\\x01\\x09\\x01\\xa1\\x00\\x05\\x09\\x19\\x01\\x29\\x05\\x15\\x00\\x25\\x01\\x95\\x05\\x75\\x01\\x81\\x02\\x95\\x01\\x75\\x03\\x81\\x01\\x05\\x01\\x09\\x30\\x09\\x31\\x09\\x38\\x15\\x81\\x25\\x7F\\x75\\x08\\x95\\x03\\x81\\x06\\xc0\\xc0 > report_desc
 
 
cd $GADGET_PATH
echo 0x046a > idVendor
echo 0x002a > idProduct
 
cd $STRINGS_PATH
echo "tejado" > manufacturer
echo "Android USB Gadget" > product
echo "42" > serialnumber
 
cd $CONFIG_PATH
echo 120 > MaxPower
echo "Configuration" > strings/0x409/configuration
 
ln -s ${GADGET_PATH}/functions/hid.keyboard $CONFIG_PATH/hid.keyboard
ln -s ${GADGET_PATH}/functions/hid.mouse $CONFIG_PATH/hid.mouse
 
ROOT (stdout) - 
ROOT (stderr) - : <stdin>[53]: can't create strings/0x409/configuration: No such file or directory
```

As you can see I get the following error `ROOT (stderr) - : <stdin>[53]: can't create strings/0x409/configuration: No such file or directory`

Which seems normal as this directory is not created by the script.
Indeed the `strings` folder created is `$GADGET_PATH/strings/0x409/` which corresponds to `/config/usb_gadget/keyboard/strings/0x409/` different from `$CONFIG_PATH/strings/0x409/configuration` which resolves to `/config/usb_gadget/keyboard/configs/c.1//strings/0x409/configuration`

Is there a quick fix for this issue?

Thanks in advance

tejado, 2023-01-12:
Hi,

yes, this was fixed with PR https://github.com/tejado/android-usb-gadget/pull/40
But it isn't yet released in F-Droid. 

Anyway, I guess this is not a blocking issue. So it should work anyway.

SF73, 2023-01-12:
Oh OK.
Well if this issue is not related, I don't know why I can't get it work...

On windows 10, I get a USB device with an error (translated to english)

```
Windows has stopped this device because it has reported problems. (code 43)

A request for the USB device descriptor failed
```

SF73, 2023-01-12:
Mmmh... I don't know if it what you're looking for but I get these two events

```xml
<Event xmlns="http://schemas.microsoft.com/win/2004/08/events/event">
<System>
  <Provider Name="Microsoft-Windows-Kernel-PnP" Guid="{9c205a39-1250-487d-abd7-e831c6290539}" /> 
  <EventID>400</EventID> 
  <Version>0</Version> 
  <Level>4</Level> 
  <Task>0</Task> 
  <Opcode>0</Opcode> 
  <Keywords>0x4000000000000000</Keywords> 
  <TimeCreated SystemTime="2023-01-12T14:11:33.9726437Z" /> 
  <EventRecordID>8756</EventRecordID> 
  <Correlation /> 
  <Execution ProcessID="4" ThreadID="19300" /> 
  <Channel>Microsoft-Windows-Kernel-PnP/Configuration</Channel> 
  <Computer>DESKTOP-FRN3BHQ</Computer> 
  <Security UserID="S-1-5-18" /> 
  </System>
<EventData>
  <Data Name="DeviceInstanceId">USB\VID_0000&PID_0002\5&150ee2ad&0&4</Data> 
  <Data Name="DriverName">usb.inf</Data> 
  <Data Name="ClassGuid">{36fc9e60-c465-11cf-8056-444553540000}</Data> 
  <Data Name="DriverDate">06/21/2006</Data> 
  <Data Name="DriverVersion">10.0.19041.1949</Data> 
  <Data Name="DriverProvider">Microsoft</Data> 
  <Data Name="DriverInbox">true</Data> 
  <Data Name="DriverSection">BADDEVICE.Dev.NT</Data> 
  <Data Name="DriverRank">0xff0000</Data> 
  <Data Name="MatchingDeviceId">USB\DEVICE_DESCRIPTOR_FAILURE</Data> 
  <Data Name="OutrankedDrivers">usb.inf:USB\DEVICE_DESCRIPTOR_FAILURE:00FF2000</Data> 
  <Data Name="DeviceUpdated">false</Data> 
  <Data Name="Status">0x0</Data> 
  <Data Name="ParentDeviceInstanceId">USB\ROOT_HUB30\4&29f2cc3c&0&0</Data> 
  </EventData>
  </Event>
  ```
  And
  ```xml
  <Event xmlns="http://schemas.microsoft.com/win/2004/08/events/event">
<System>
  <Provider Name="Microsoft-Windows-Kernel-PnP" Guid="{9c205a39-1250-487d-abd7-e831c6290539}" /> 
  <EventID>411</EventID> 
  <Version>0</Version> 
  <Level>2</Level> 
  <Task>0</Task> 
  <Opcode>0</Opcode> 
  <Keywords>0x4000000000000000</Keywords> 
  <TimeCreated SystemTime="2023-01-12T14:11:33.9802063Z" /> 
  <EventRecordID>8757</EventRecordID> 
  <Correlation /> 
  <Execution ProcessID="4" ThreadID="19300" /> 
  <Channel>Microsoft-Windows-Kernel-PnP/Configuration</Channel> 
  <Computer>DESKTOP-FRN3BHQ</Computer> 
  <Security UserID="S-1-5-18" /> 
  </System>
<EventData>
  <Data Name="DeviceInstanceId">USB\VID_0000&PID_0002\5&150ee2ad&0&4</Data> 
  <Data Name="DriverName">usb.inf</Data> 
  <Data Name="ClassGuid">{36fc9e60-c465-11cf-8056-444553540000}</Data> 
  <Data Name="ServiceName" /> 
  <Data Name="LowerFilters" /> 
  <Data Name="UpperFilters" /> 
  <Data Name="Problem">0x0</Data> 
  <Data Name="Status">0xc00000e5</Data> 
  </EventData>
  </Event>
  ```

tejado, 2023-01-16:
What ROM are you using and which kernel version has your ROM?

SF73, 2023-02-08:
> What ROM are you using and which kernel version has your ROM?

I'm using Android 13 (TQ1A.221205.011). The kernel version is 5.10.107
So apparently I can edit the default gadget (g1 in my case) and add functions. I tried keyboard and it is working as expected.
However, if I try to create a new gadget I get this error.