tejado/android-usb-gadget #43: Kernel 5.10 strange behaviour

iodn, 2023-01-15:
I have been using usb-gadget tool for a while on my Xiaomi 11 Lite (Kernel 4.14) and no problem. 
But I recently purchased a Pixel 6a, /dev/hidg0 and /dev/hidg1 are present and enabled.

The strange thing is that when I execute a command on /dev/hidg0 <br>e.g. 
```bash
echo h i | ./hid-gadget /dev/hidg0 keyboard
```
The output is reflected on the mobile phone itself instead of the device that should be connected (PC or another device ..).

Do you have any idea why it behaves this way?

tejado, 2023-01-16:
Hi,

I don't have any phone with kernel 5.1 (yet), so I can't test it. I might try this on a Pi. I don't have any clue - sorry.
What is the value of getprop sys.usb.controller ?
What ROM are you using? Stock? And which release exactly? Then I can check if there is something new in the kernel.

Tjado

iodn, 2023-01-16:
Okok I see, no problem.

> What is the value of getprop sys.usb.controller ?
```
bluejay:/ # getprop sys.usb.controller
dummy_udc.0
```

> What ROM are you using? Stock? And which release exactly? Then I can check if there is something new in the kernel.

Well I started with the official Google rom from this month (Jan. 2023, https://support.google.com/pixelphone/thread/195623748/google-pixel-update-january-2023?hl=en) 

Then I realized that usb-gadget tool didn't work, so I installed a custom rom (GrapheneOS) (https://grapheneos.org/releases#bluejay-stable) thinking that I might have better luck but the result was the same.

I've also thought about downgrading the kernel to try, but I'm still not sure how to do this on Android and I need to do more research.

tejado, 2023-01-16:
> > What is the value of getprop sys.usb.controller ?
> 
> ```
> bluejay:/ # getprop sys.usb.controller
> dummy_udc.0
> ```

This is a dummy USB controller and not the actual one. I guess this is the issue.
Can you look whats inside the directory /sys/class/udc and as well activate ADB and see what value the default USB gadget has, e.g. by executing the command '/config/usb_gadget/*/UDC' ?

iodn, 2023-01-16:
```
bluejay:/ # ls /sys/class/udc/
11110000.dwc3  dummy_udc.0
bluejay:/ # cat /sys/class/udc/dummy_udc.0/uevent
USB_UDC_NAME=dummy_udc
bluejay:/ # cat /config/usb_gadget/*/UDC
11110000.dwc3

```

tejado, 2023-01-16:
> ```
> bluejay:/ # ls /sys/class/udc/
> 11110000.dwc3  dummy_udc.0
> bluejay:/ # cat /sys/class/udc/dummy_udc.0/uevent
> USB_UDC_NAME=dummy_udc
> bluejay:/ # cat /config/usb_gadget/*/UDC
> 11110000.dwc3
> ```

Great, then use USB Gadget tool as usual. But if you want to activate the respective gadget, e.g. MouseKeyboard, please test the following:

echo "" > /config/usb_gadget/*/UDC
echo 11110000.dwc3 > /config/usb_gadget/MouseKeyboard/UDC

So I guess I need to adjust the controller detection in USB Gadget tool.

iodn, 2023-01-16:
Hmm, every time I try to overwrite the `/config/usb_gadget/MouseKeyboard/UDC` file it comes back to the initial `dummy_udc.0`

```
bluejay:/ # vim /config/usb_gadget/MouseKeyboard/UDC                                                                  
bluejay:/ # cat /config/usb_gadget/MouseKeyboard/UDC                                                                  
dummy_udc.0
bluejay:/ # echo 11110000.dwc3 > /config/usb_gadget/MouseKeyboard/UDC                                                 
1|bluejay:/ # cat /config/usb_gadget/MouseKeyboard/UDC                                                                
dummy_udc.0

```

tejado, 2023-01-16:
What about the other gadget, the default one? Does it still has 11110000.dwc3 in it? Can you also paste the kernel log (dmesg) if you are writing to UDC? Thanks.
Please paste these logs on pastebin - not directly here as it will otherwise too much text.

iodn, 2023-01-16:
> What about the other gadget, the default one? Does it still has 11110000.dwc3 in it?

```
bluejay:/ # echo 11110000.dwc3 > /config/usb_gadget/MouseKeyboard/UDC                                                 
1|bluejay:/ # cat /config/usb_gadget/MouseKeyboard/UDC                                                                
dummy_udc.0
bluejay:/ # cat /config/usb_gadget/g1/UDC                                                                             
11110000.dwc3
```
The file was too big for Pastebin (You have exceeded the maximum size of 512 kilobytes per Paste.), I uploaded it on controlc.com

[https://controlc.com/cafc4b44](https://controlc.com/cafc4b44)

tejado, 2023-01-16:
Thank you - can't see anything in the log which relates to this.
I will need to investigate - unfortunately, I don't have any device which this kernel.

If you can find out why this is happeing, it would be great.

iodn, 2023-01-17:
Update, when I setprop `ro.boot.usbcontroller` and `sys.usb.controller` it seems to work and could overwrite the `/config/usb_gadget/MouseKeyboard/UDC` file but my computer still doesn't recognize the mobile phone as HID device and nothing happen when I try to execute some commands.

```
bluejay:/ # cat /config/usb_gadget/MouseKeyboard/UDC
dummy_udc.0
bluejay:/ # setprop ro.boot.usbcontroller 11110000.dwc3
bluejay:/ # setprop sys.usb.controller 11110000.dwc3
bluejay:/ # ##Re-enable MouseKeyboard gadget from the app
bluejay:/ # cat /config/usb_gadget/MouseKeyboard/UDC
11110000.dwc3
```
[image]

Another strange fact, if I load `hid.keyboard` directly from the path gadget by default `/config/usb_gadget/g1/` it works halfway. For some reasons my windows machine still doesn't recognize the mobile phone as HID device but when I plugged it into my Linux box the HID payload I coded kind of work but all the characters are in uppercase and the special characters like slash `/` are replaced by interrogation marks `?`.

tejado, 2023-01-17:
Okay, interessting. Thanks a lot for the information. I will hopefully get a device with such a Kernel in the next days, so I can look more into that.

Under Windows, do you get the same error in the Device Manager as the following issue? https://github.com/tejado/android-usb-gadget/issues/42

If you are using hid-gadget tool, I can't help you there as I'm not using it. Try [Authorizer](https://github.com/tejado/Authorizer).

iodn, 2023-01-17:
> Okay, interesting. Thanks a lot for the information. I will hopefully get a device with such a Kernel in the next days, so I can look more into that.

Oh so cool, that's good to hear then!

> Under Windows, do you get the same error in the Device Manager as the following issue? https://github.com/tejado/android-usb-gadget/issues/42

No, there is no error message, the phone is just not recognized at all by the computer.
 > If you are using hid-gadget tool, I can't help you there as I'm not using it. Try [Authorizer](https://github.com/tejado/Authorizer).

Yes, I know but my test script works perfectly on two other phones, so it should also work in theory with the Pixel 6. But thanks for the info, I'll try Authorizer.

SF73, 2023-05-08:
Hey, any news concerning this issue?
So far, it seems possible to edit the default gadget (g1) but not create one from scratch.

MikeDev101, 2023-08-21:
Having a similar issue on the Pixel 6 (oriole). I edited the props as mentioned in https://github.com/tejado/android-usb-gadget/issues/43#issuecomment-1384876720 but I couldn't get any devices to detect USB HID mode.

```
ROOT - Execute command: for dir in /config/usb_gadget/*/; do echo GADGET_PATH=$dir; cd $dir/configs/; echo CONFIG_PATH="$dir/configs/`ls -1 | head -1`/"; cd $dir; if [ "$?" -ne "0" ]; then echo "Error - not able to change dir to $dir... exit"; exit 1; fi; echo UDC=$(cat UDC); find ./configs/ -type l -exec sh -c 'echo FUNCTIONS_ACTIVE=$(basename $(readlink "$@"))' _ {} \;; for f in ./functions/*/; do echo FUNCTIONS=$(basename $f); done; cd ./strings/0x409/; for vars in *; do echo ${vars}=$(cat $vars); done; echo "============="; done; 

ROOT (stdout) - GADGET_PATH=/config/usb_gadget/g1/
CONFIG_PATH=/config/usb_gadget/g1//configs/b.1/
UDC=dummy_udc.0
FUNCTIONS_ACTIVE=ffs.mtp
FUNCTIONS=accessory.gs2
FUNCTIONS=acm.gs6
FUNCTIONS=acm.uwb0
FUNCTIONS=audio_source.gs3
FUNCTIONS=dm.gs7
FUNCTIONS=etr_miu.gs11
FUNCTIONS=ffs.adb
FUNCTIONS=ffs.mtp
FUNCTIONS=ffs.ptp
FUNCTIONS=midi.gs5
FUNCTIONS=ncm.gs9
FUNCTIONS=rndis.gs4
manufacturer=Google
product=Pixel 6
serialnumber=1C181FDF6001BX
=============
GADGET_PATH=/config/usb_gadget/keyboard/
CONFIG_PATH=/config/usb_gadget/keyboard//configs/c.1/
UDC=11110000.dwc3
FUNCTIONS_ACTIVE=hid.mouse
FUNCTIONS_ACTIVE=hid.keyboard
FUNCTIONS=hid.keyboard
FUNCTIONS=hid.mouse
manufacturer=tejado
product=Android USB Gadget
serialnumber=42
=============

ROOT (stderr) - 
GadgetShellApi - Gadget Path is valid
ROOT - Execute command: echo "" > /config/usb_gadget/keyboard//UDC

ROOT (stdout) - 
ROOT (stderr) - 
ROOT - Execute command: for dir in /config/usb_gadget/*/; do echo GADGET_PATH=$dir; cd $dir/configs/; echo CONFIG_PATH="$dir/configs/`ls -1 | head -1`/"; cd $dir; if [ "$?" -ne "0" ]; then echo "Error - not able to change dir to $dir... exit"; exit 1; fi; echo UDC=$(cat UDC); find ./configs/ -type l -exec sh -c 'echo FUNCTIONS_ACTIVE=$(basename $(readlink "$@"))' _ {} \;; for f in ./functions/*/; do echo FUNCTIONS=$(basename $f); done; cd ./strings/0x409/; for vars in *; do echo ${vars}=$(cat $vars); done; echo "============="; done; 

ROOT (stdout) - GADGET_PATH=/config/usb_gadget/g1/
CONFIG_PATH=/config/usb_gadget/g1//configs/b.1/
UDC=dummy_udc.0
FUNCTIONS_ACTIVE=ffs.mtp
FUNCTIONS=accessory.gs2
FUNCTIONS=acm.gs6
FUNCTIONS=acm.uwb0
FUNCTIONS=audio_source.gs3
FUNCTIONS=dm.gs7
FUNCTIONS=etr_miu.gs11
FUNCTIONS=ffs.adb
FUNCTIONS=ffs.mtp
FUNCTIONS=ffs.ptp
FUNCTIONS=midi.gs5
FUNCTIONS=ncm.gs9
FUNCTIONS=rndis.gs4
manufacturer=Google
product=Pixel 6
serialnumber=1C181FDF6001BX
=============
GADGET_PATH=/config/usb_gadget/keyboard/
CONFIG_PATH=/config/usb_gadget/keyboard//configs/c.1/
UDC=
FUNCTIONS_ACTIVE=hid.mouse
FUNCTIONS_ACTIVE=hid.keyboard
[... 472 lines cut ...]
FUNCTIONS=etr_miu.gs11
FUNCTIONS=ffs.adb
FUNCTIONS=ffs.mtp
FUNCTIONS=ffs.ptp
FUNCTIONS=midi.gs5
FUNCTIONS=ncm.gs9
FUNCTIONS=rndis.gs4
manufacturer=Google
product=Pixel 6
serialnumber=1C181FDF6001BX
=============
GADGET_PATH=/config/usb_gadget/keyboard/
CONFIG_PATH=/config/usb_gadget/keyboard//configs/c.1/
UDC=
FUNCTIONS_ACTIVE=hid.mouse
FUNCTIONS_ACTIVE=hid.keyboard
FUNCTIONS=hid.keyboard
FUNCTIONS=hid.mouse
manufacturer=tejado
product=Android USB Gadget
serialnumber=42
=============

ROOT (stderr) - 
```