tejado/android-usb-gadget #45: Phone 100% reboots when writing keyboard data into /dev/hidg1 after USB unplugged

WuDi-ZhanShen, 2023-03-04:
I have successfully made an APP to change my Android to a USB Keyoard by writing data into /dev/hidg1 after using GSB Gadget Tool ----add function --- Keyboard !! That's really cool. The only question is why my Android 100% reboots automaticly when i unplug the USB from my computer and then write a keyboard data to /dev/hidg1.

And i want adding new USB function "GamePad" into USB Gadget Tool !! Thanks a lot !!

tejado, 2023-03-05:
Can you try to add a new gadget and NOT a new function to an existing gadget? Depending on the phone vendor, there might be some incompatibilities.
What phone and OS (stock or custom rom, Android Version, Kernel version) are you using?

Adding GamePad should be really simple. We just need to have the respective descriptor for the GamePad you want.

WuDi-ZhanShen, 2023-03-06:
I tried adding a new gadget 'Keyboard&Mouse' , and there's two new higds appeared in /dev/hidg2 and /dev/hidg3. However I can't writing data into these two hidgs. When i try to write, there will be an IOException Error Message:

Error!

write failed: errno 108 (Cannot send after transport endpoint shutdown)

I'm confused because after that I change to the ordinally gadget and add a Keyboard function and writing data into /dev/hidg1, and there is no Error Message and USB keyboard works well.

My phone is Redmi Note 11T Pro, Android 12 

KERNEL_VERSION=5.10.66
CONFIG_PCI_ENDPOINT_CONFIGFS=NOT_SET
CONFIG_USB_CONFIGFS=y
CONFIG_USB_CONFIGFS_UEVENT=y
CONFIG_USB_CONFIGFS_SERIAL=y
CONFIG_USB_CONFIGFS_ACM=y
CONFIG_USB_CONFIGFS_OBEX=NOT_SET
CONFIG_USB_CONFIGFS_NCM=y
CONFIG_USB_CONFIGFS_ECM=y
CONFIG_USB_CONFIGFS_ECM_SUBSET=NOT_SET
CONFIG_USB_CONFIGFS_RNDIS=y
CONFIG_USB_CONFIGFS_EEM=y
CONFIG_USB_CONFIGFS_MASS_STORAGE=y
CONFIG_USB_CONFIGFS_F_LB_SS=NOT_SET
CONFIG_USB_CONFIGFS_F_FS=y
CONFIG_USB_CONFIGFS_F_ACC=y
CONFIG_USB_CONFIGFS_F_AUDIO_SRC=y
CONFIG_USB_CONFIGFS_F_UAC1=NOT_SET
CONFIG_USB_CONFIGFS_F_UAC1_LEGACY=NOT_SET
CONFIG_USB_CONFIGFS_F_UAC2=y
CONFIG_USB_CONFIGFS_F_MIDI=y
CONFIG_USB_CONFIGFS_F_HID=y
CONFIG_USB_CONFIGFS_F_UVC=y
CONFIG_USB_CONFIGFS_F_PRINTER=NOT_SET
CONFIG_IIO_CONFIGFS=NOT_SET
CONFIG_CONFIGFS_FS=y

WuDi-ZhanShen, 2025-11-12:
> 时隔多年，打扰了。
>  同红米note11tp，请问您是怎么让键盘在这台手机上正常工作的？
>  我尝试了将/dev/hidg1&amp;2重命名为hidg0&amp;1，键盘软件就不会报错，但仍然无法被Windows识别；
>  也尝试将.66内核的USB源码更新到5.10 lts最新，仍然无效；
>  也尝试了老哥你仓库里的那个软件，按你写的步骤来，还是无效..
>  求老哥指教一二😭
>
> Screenshot_2025-11-12-17-59-27-383_com.emu.anyUSB.jpg (view on web)

是否hidg1和2设置为键盘和鼠标的顺序反了呢

Kissycat, 2025-11-13:
找到问题[所在](https://github.com/Rohail33/Realking_xiaomi_xaga/blob/267394110efabd659eda20183a41f386fbb552c0/include/linux/hid.h#L673)了，米米内核里这段不知道从哪来的代码害的，但是把1改成0就好了...