tejado/android-usb-gadget #46: compilation error

francoisnicolas, 2023-10-18:
1)  get this error  :  tjado.usbgadge: open libmigui.so failed! dlopen - dlopen failed: library "libmigui.so" not found

compiled with Android Studio on a windows PC
version : Android Studio Electric Eel | 2022.1.1
Build #AI-221.6008.13.2211.9477386, built on January 11, 2023
Runtime version: 11.0.15+0-b2043.56-8887301 amd64
VM: OpenJDK 64-Bit Server VM by JetBrains s.r.o.
Windows 11 10.0
GC: G1 Young Generation, G1 Old Generation
Memory: 4096M
Cores: 8
Registry:
    external.system.auto.import.disabled=true
    ide.text.editor.with.preview.show.floating.toolbar=false

2) Is it necessary to root the smartphone, in order to use this app ?

tejado, 2023-10-22:
1) Do you have any stack trace related to that error? I don't know what libmigui.so is and why it is related to compilation of my app.

2) Requirements are stated in the readme: "USB Gadget Tool requires root permissions and a Kernel with ConfigFS support."

francoisnicolas, 2023-10-22:
E/AndroidRuntime: FATAL EXCEPTION: main
    Process: net.tjado.usbgadget, PID: 29091
    java.lang.NullPointerException: null cannot be cast to non-null type kotlin.String
        at net.tjado.usbgadget.GadgetShellApi.updateGadgetData$lambda$1(GadgetShellApi.kt:68)
        at net.tjado.usbgadget.GadgetShellApi.$r8$lambda$ZXtX09HUF-P_VvkzGmTntqu3KsU(Unknown Source:0)
        at net.tjado.usbgadget.GadgetShellApi$$ExternalSyntheticLambda1.onRootTaskFinish(Unknown Source:6)
        at net.tjado.usbgadget.RootTask.onPostExecute(RootTask.kt:29)
        at net.tjado.usbgadget.RootTask.onPostExecute(RootTask.kt:5)
        at net.tjado.usbgadget.CoroutineTask$execute$1.invokeSuspend(CoroutineTask.kt:28)
        at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
        at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:106)
        at android.os.Handler.handleCallback(Handler.java:942)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at android.os.Looper.loopOnce(Looper.java:226)
        at android.os.Looper.loop(Looper.java:313)
        at android.app.ActivityThread.main(ActivityThread.java:8762)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:604)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1067)
    	Suppressed: kotlinx.coroutines.DiagnosticCoroutineContextException: [StandaloneCoroutine{Cancelling}@ab32734, Dispatchers.Main]
I/Process: Sending signal. PID: 29091 SIG: 9

francoisnicolas, 2023-10-22:
... 
I hadn't looked closely at the error message, it's a runtime error message since my smartphone is not rooted. I thought it was a compilation error