tejado/android-usb-gadget #47: Can it be accessed by terminal emulators like termux ?

rexackermann, 2023-12-31:


Sucareto, 2024-02-22:
Turn on hid
`ls /sys/class/udc > /config/usb_gadget/keyboard/UDC`
Turn off hid
`echo > /config/usb_gadget/keyboard/UDC`

rexackermann, 2024-02-26:
> Turn on hid `ls /sys/class/udc > /config/usb_gadget/keyboard/UDC` Turn off hid `echo > /config/usb_gadget/keyboard/UDC`

Thanks a lot for the reply but it is not working.
The keyboard dir itself does not exist.

Sucareto, 2024-02-26:
> > Turn on hid `ls /sys/class/udc > /config/usb_gadget/keyboard/UDC` Turn off hid `echo > /config/usb_gadget/keyboard/UDC`
> 
> Thanks a lot for the reply but it is not working. The keyboard dir itself does not exist.

The keyboard directory is created by the android-usb-gadget,You first need to create the Mouse & Keyboard Gadget in app.
If you want to complete gadget creation in termux, you can refer to the script [here](https://github.com/tejado/android-usb-gadget/tree/master/app/src/main/assets).
By the way, I wrote a complete script [here](https://gist.github.com/Sucareto/f8b9b1d0594dd87ba1318db549282b67).